zephyrproject-rtos/zephyr #5554: Support maintaining drivers and other Zephyr libraries out-of-tree

SebastianBoe, 2018-01-04:
# Use-case
Users want to maintain Zephyr components out-of-tree, boards, arch's, drivers etc.

This github issue tracks the use-case of maintaining a driver out-of-tree.

# Workaround today
When an application needs a custom driver that is not supported by Zephyr it would
have to maintain it in a fork of Zephyr that contained the driver. But common use-cases
such as this should not require a Zephyr fork.

# Solution
Perhaps one way to support this would be for Zephyr to support hooking in external "kernel" CMakeLists.txt code. Something like this:
https://gist.github.com/SebastianBoe/59e611325958e98fe4f0fb77a8c69d63

SebastianBoe, 2018-01-11:
Strange, was not able to reproduce the issue.

sample: zephyr/samples/hello_world
Board: nrf52_pca10040
patch: https://github.com/SebastianBoe/zephyr/commit/26e947dec325ce14ec0b506ffe40179ce0b2b52e

After building I was able to observe that devconfig was populated with an entry for the 'app' driver.

> devconfig       0x000000000000a3a0       0x84
>                 0x000000000000a3a0                __devconfig_start = .
>  *(.devconfig.*)
>  .devconfig.init
>                 0x000000000000a3a0        0xc libapp.a(entropy_nrf5.c.obj)
>  .devconfig.init
>                 0x000000000000a3ac        0xc zephyr/libzephyr.a(soc.c.obj)
>  .devconfig.init
> 

Didn't actually test that the driver was functional, but the original report was that there was no entry in devconfig ...

pboettch, 2018-03-29:
I'm back on this link-problem and have found another case which creates a problem: libraries added at application-level which are using functions and features from zephyr which are not used by zephyr itself.

```CMake

include($ENV{ZEPHYR_BASE}/cmake/app/boilerplate.cmake NO_POLICY_SCOPE)
project(NONE)

# bad example as code.c could be added to app - imagine an imported library or a library from a sub-directories which is there because of code-structure
add_library(lib code.c) # uses pipes for example 

target_sources(app
    PRIVATE
        main.c)
target_link_libraries(app lib)
```
If now the zephyr-part the (part which is linked within `--start-group` and `--end-group`)  is not using pipes the linker throws away, for example, `_impl_k_pipe_put`. As `lib` is using it we get:

```
build/zephyr/include/generated/syscalls/kernel.h:105: undefined reference to `_impl_k_pipe_put'
build/zephyr/include/generated/syscalls/kernel.h:105: undefined reference to `_impl_k_pipe_put'
lib/liblib.a(code.c): In function `k_pipe_init':
build/zephyr/include/generated/syscalls/kernel.h:103: undefined reference to `_impl_k_pipe_init'
```

One solution is add something whichs add later added libraries to the link-group via generator-expressions. I was unable to come up with a working solutions using cmake-targets. What works are filenames. I created a custom property to which I added filenames at the same time when calling `target_link_libraries()` in app's CMakeLists.txt. However, knowing the filename is not always easy.

Then I found several discussions talking about how to handle circular dependencies of libraries (not really ours case but the problem is similar). CMake has that [LINK_INTERFACE_MULTIPLICITY](https://cmake.org/cmake/help/v3.8/prop_tgt/LINK_INTERFACE_MULTIPLICITY.html#prop_tgt:LINK_INTERFACE_MULTIPLICITY) variable which adds the dependent static libraries multiple times to the linker-line. This looks promising to me, but I don't if there is a special reason for this project to use `--start-group` and `--end-group`.

What would you say?

SebastianBoe, 2018-04-04:
Hi,

will get this cleared up by the end of the week.

pboettch, 2018-04-04:
I'm available as well for discussions (elsewhere than here or by mail if needed). I'd love to have a nice solution.

SebastianBoe, 2018-04-06:
I can confirm that out-of-tree drivers are not supported yet because drivers use system calls and the system call infrastructure is currently assuming that all system call declarations are located in 
zephyr/include.

Offending line:
https://github.com/zephyrproject-rtos/zephyr/blob/master/CMakeLists.txt#L343

SebastianBoe, 2018-04-06:
@pboettch : I will create a new issue for this observation/bugreport:
https://github.com/zephyrproject-rtos/zephyr/issues/5554#issuecomment-377183960

pboettch, 2018-04-06:
OK. Note that syscalls haven't been a problem in my environment (yet?!). Out-of-tree-drivers and -libraries are creating link problems because they are not in the link-group.

andrewboie, 2018-05-08:
> I can confirm that out-of-tree drivers are not supported yet because drivers use system calls and the system call infrastructure is currently assuming that all system call declarations are located in
zephyr/include.

This is not really correct.
Driver system calls are declared at the subsystem level and not the actual drivers themselves.
I do not see a use-case for having out-of-tree driver subsystems or out of tree system calls. You should still be able to create out-of-tree drivers without making new syscalls.

SebastianBoe, 2018-05-09:
"I do not see a use-case for having out-of-tree driver subsystems or out of tree system calls."

It took me a while to think of one as well. But here is one:

An IC-vendor subscribes to the micro-repo methodology of version control.
Also, they are developing HW/SW for a driver-type that is secret until released
and happens to not be supported by Zephyr yet.

Then they need new syscall declarations, because Zephyr does not support the HW yet,
and they can't declare them in-tree, because they wish to maintain things out-of-tree.

andrewboie, 2018-05-09:
> Also, they are developing HW/SW for a driver-type that is secret until released
and happens to not be supported by Zephyr yet.

nvlsianpu, 2018-05-14:
Actually it in not hard to describe another user case (which i encountered many time in my career): An customer have a very specific requirement for a driver - so it is required to provide a user-case-driver by him. Because of business reason the customer can't even show this driver to the zephyr community. I would be pointless to keep this driver on a zephyr branch - it would be better to have it in separate repo (at last it would mitigate problem while user is intended to synchronize with the zephyr upstream). I also give protection against sending this driver to the community accidentally (while user might want to send some other contribution).

closed 2018-05-31 by SebastianBoe