zephyrproject-rtos/zephyr #5589: Issue with using generic gcc cross compile with cmake

nullr0ute, 2018-01-08:
I've tried both the 1.10 branch and master with this, but I can't seem to get zephyr to build with generic distro cross compilers when using the SDK less instructions here http://docs.zephyrproject.org/getting_started/getting_started.html#sdkless-builds like I could with < 1.10 (details of older here: https://nullr0ute.com/2017/03/getting-started-with-zephyr-on-fedora/ )

It seems related to the fixes in pr #4918. 

Looking through the sdkless instructions and cmake/toolchain-gcc.cmake I believe I should be able to set the CROSS_COMPILE env variable and the following bits:
<pre>
export CROSS_COMPILE="/usr/bin/arm-linux-gnu-"
unset ZEPHYR_GCC_VARIANT
unset ZEPHYR_SDK_INSTALL_DIR
source zephyr-env.sh

cd samples/hello_world/
mkdir build && cd build
cmake -DBOARD=arduino_due ..
</pre>

But that gives the following error:
<pre>
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.3", minimum required is "3.4") 
CMake Error at /home/perobins/code/zephyr/cmake/app/boilerplate.cmake:102 (message):
  ZEPHYR_GCC_VARIANT not set
Call Stack (most recent call first):
  CMakeLists.txt:3 (include)
</pre>

If you set  ZEPHYR_GCC_VARIANT to gcc which I think should then use the cmake/toolchain-gcc.cmake settings like below:
<pre>
export CROSS_COMPILE="/usr/bin/arm-linux-gnu-"
export ZEPHYR_GCC_VARIANT=gcc
unset ZEPHYR_SDK_INSTALL_DIR
source zephyr-env.sh

cd samples/hello_world/
mkdir build && cd build
cmake -DBOARD=arduino_due ..
</pre>

You get the following error:
<pre>
Zephyr version: 1.10.99
Using /home/perobins/code/zephyr/boards/arm/arduino_due/arduino_due_defconfig as base
Merging /home/perobins/code/zephyr/samples/hello_world/prj.conf
Merging /home/perobins/code/zephyr/tests/include/test.config
#
# configuration written to /home/perobins/code/zephyr/samples/hello_world/build/zephyr/.config
#
-- Generating zephyr/include/generated/autoconf.h
CMake Error at /home/perobins/code/zephyr/cmake/extensions.cmake:964 (message):
  No such file or directory: CMAKE_READELF: 'readelf'
Call Stack (most recent call first):
  /home/perobins/code/zephyr/cmake/toolchain-gcc.cmake:14 (assert_exists)
  /home/perobins/code/zephyr/cmake/toolchain.cmake:15 (include)
  /home/perobins/code/zephyr/cmake/app/boilerplate.cmake:229 (include)
  CMakeLists.txt:3 (include)
</pre>

The error above seems to indicate that the ${CROSS_COMPILE} variable that's referenced in cmake/toolchain-gcc.cmake is being unset/not sources or something, all the referenced files there when pre pended with CROSS_COMPILE="/usr/bin/arm-linux-gnu-" exist.

My cmake skills aren't really good enough to get me any further than the above.

SebastianBoe, 2018-01-08:
Hi, thank you for the detailed feedback.

I expect this to be fixed when #5594 is merged.
But with one change, instead of

export CROSS_COMPILE="/usr/bin/arm-linux-gnu-"
export ZEPHYR_GCC_VARIANT=gcc

do

export CROSS_COMPILE="/usr/bin/arm-linux-gnu-"
export ZEPHYR_GCC_VARIANT=cross-compile

or

export CROSS_COMPILE="/usr/bin/arm-linux-gnu-"
unset ZEPHYR_GCC_VARIANT

nullr0ute, 2018-01-09:
Initial testing (not extensive) looks good. It certainly gets past the issue I was seeing above and looks good. I'll do some more extensive testing when I get some spare cycles but this LGTM

PR #5594 (2018-01-12): Support cross compile
CROSS_COMPILE is a KBuild feature that was dropped during the CMake
migration. It is now re-introduced. Documentation for it is still
lacking, but at least it now behaves as expected.

This fixes #5589
closed 2018-01-12 by nashif