zephyrproject-rtos/zephyr #5603: Bluetooth debug is hardcoded with level 4

nashif, 2018-01-08:
In subsys/bluetooth/common/log.h

`#define SYS_LOG_LEVEL SYS_LOG_LEVEL_DEBUG`

So it is either DEBUG or nothing in this case.. 

It should be possible to set to other levels.

jhedberg, 2018-01-09:
This is not really a bug, since the debug vs no-debug logging is decided on a per-submodule basis (see the condition that BT_DBG expands to). Since it's the per-submodule Kconfig switches that decide whether debug logs get output the SYS_LOG_LEVEL must always be set to SYS_LOG_LEVEL_DEBUG. Otherwise when the condition in BT_DBG evaluates to true the SYS_LOG_DBG call would not output anything.

nashif, 2018-01-13:
Where exactly do you set the log level for bluetooth to be 1 or 2 or 3 for example? I only see this being set as debug with no way for me to turn off debugging and get errors only for example.

jhedberg, 2018-01-14:
Bluetooth debug logs are not opt-out. They're opt-in. You get errors and warnings by enabling CONFIG_BT_DEBUG_LOG=y. That will not log any BT_DBG() calls however. To get those, you need to enabled the appropriate CONFIG_BT_DEBUG_* option for the sub-module that you want debug logs for. Note that if you're writing new code and there's no CONFIG_BLUETOOTH_DEBUG_* option for your new sub-module, then BT_DBG() does show up unconditionally, however all existing Bluetooth sub-modules have their own CONFIG_BT_DEBUG_* option which needs to be explicitly enable to get the BT_BG() logs.

jhedberg, 2018-01-14:
Btw, I realize the name of CONFIG_BT_DEBUG_LOG may be a bit misleading since it does not enable BT_DBG(). Perhaps a better name would be simply CONFIG_BT_LOG.

nashif, 2018-01-14:
still does not answer my question, how do you reduce the log level for bluetooth? i.e. get only INFO and below (BT_INFO, BT_ERR, ..), not debug messages (BT_DBG)?

jhedberg, 2018-01-14:
> still does not answer my question, how do you reduce the log level for bluetooth? i.e. get only INFO and below (BT_INFO, BT_ERR, ..), not debug messages (BT_DBG)?

That's what I thought I answered (apparently not clearly enough). To get BT_INFO, BT_WARN and BT_ERR, but not BT_DBG, you'd set CONFIG_BT_DEBUG_LOG=y and not set any of the submodule-specific CONFIG_BT_DEBUG_* options (such as CONFIG_BT_DEBUG_DRIVER or CONFIG_BT_DEBUG_HCI_CORE)

jhedberg, 2018-01-14:
Take a look what BT_DBG expands to. It will only log if BT_DBG_ENABLED is set. And this variable gets set in each sub-module before including log.h like this:

```
#define BT_DBG_ENABLED IS_ENABLED(CONFIG_BT_DEBUG_HCI_CORE)
#include "common/log.h"
```

nashif, 2018-01-14:
right, so I can choose to debug or not to debug, but I can't set the log level anywhere to for example "errors only" (level 1), because 

`#define BT_INFO(fmt, ...) SYS_LOG_INF(fmt, ##__VA_ARGS__)`

and 

```
#if (SYS_LOG_LEVEL >= SYS_LOG_LEVEL_INFO)
#define SYS_LOG_INF(...) LOG_NO_COLOR(SYS_LOG_TAG_INF, ##__VA_ARGS__)
#endif
```

and you have set 

`#define SYS_LOG_LEVEL SYS_LOG_LEVEL_DEBUG`

so, you basically reduced the levels to debug or no debug. BT does not have a way to set the level, and you make the decision based on internal handling inside the bluetooth subsystem.

So either I missing something here, or we are talking about two different things

jhedberg, 2018-01-14:
I think what's confusing things here is that the Bluetooth logging macros map to SYS_LOG. Those mappings used to originally be directly to printk. Maybe we should have left it like that until the whole thing got converted to SYS_LOG. Note that there are other back-ends to the Bluetooth logging macros than SYS_LOG, in particular the monitor protocol, so their semantics should be the same regardless of what backend is chosen. So you can't really compare the logging levels with SYS_LOG, but there *are* three logging levels there as far as I see: no logs, info/warn/err, and info/warn/err + dbg.

nashif, 2018-01-14:
>  there are three logging levels there as far as I see: no logs, info/warn/err, and info/warn/err + dbg.

which is exactly my point, you are not following the log levels of sys_log, if someone wants to only get errors, they can't right now.

jhedberg, 2018-01-14:
Correct. And the reason is historical since these levels were defined before SYS_LOG existed.  The proper way out of this (as far as I see) would be to convert the whole thing to SYS_LOG, i.e. have SYS_LOG be the topmost layer instead of sitting under the Bluetooth macros.

jhedberg, 2018-01-14:
Before going on with the conversion there are the improvements to SYS_LOG that we've discussed, that I'd like to see merged first (without taking a position on who will do them - it might even be myself):

- Naming (shorter prefix)
- Use IS_ENABLED so format string errors/warnings are always caught
- Pass the log type/level to the backend function (needed e.g. by the Bluetooth monitor protocol)

jhedberg, 2018-01-14:
Slight correction: I don't think we need IS_ENABLED - just something that evaluates to a constant boolean expression at build-time, i.e. something you can put in an if-statement. The existing pre-prosessor conditions that test for the levels might be sufficient, i.e. just convert them from pre-prosessor macros to actual if-statements.

jhedberg, 2018-01-14:
Another thing to the list of SYS_LOG issues that needs fixing:

- Keep meta-data separate from the string to be printed, and let the backend function decide how to apply it. This includes things like the log type, color escape codes, time-stamp, etc. Right now the SYS_LOG implementation imposes many of these unconditionally up-front and doesn't give the backend function any choice in how to apply them.

Vudentz, 2018-02-19:
There are 2 ways we could apply the level, per sub-domain/file or bluetooth global. I suppose we want per sub-domain so we would likely add a kconfig entry for each sub-domain to configure the level, but we still have to decide if we leave the current DEBUG or replace with LOG_LEVEL like the other users of SYS_LOG. We can also have a master Kconfig that enables every sub-domain, though in case of net they have the level global, not per sub-domain.

As for the other improvements to the SYS_LOG I guess we will need a different issue to track them properly.

nashif, 2018-11-05:
obsolete now with new logger in place.

closed 2018-11-05 by nashif