zephyrproject-rtos/zephyr #5622: Update kconfiglib used for generating Kconfig documentation

nashif, 2018-01-10:
We do have an old version that needs to be updated and use the new one being imported into scripts/

SebastianBoe, 2018-03-20:
I spent a couple of minutes on this.

Note that this is not just a matter of updating the import. There have been backwards-incompatible changes to kconfiglib that need to be ported.

Specifically;

 Config has been renamed to Kconfig (Minor)

def get_top_level_items() does not exist any more (Unknown consequences)

https://github.com/SebastianBoe/zephyr/commit/fa404c343429da7f06454ac95c64bad10601c6ff

SebastianBoe, 2018-03-20:
This is going to break eventually. Most certainly this is a blocker for https://github.com/zephyrproject-rtos/zephyr/pull/6569

ulfalizer, 2018-03-26:
I'm looking into this now.

Another complication is that there's a custom `Symbol.rest()` function in Zephyr's Kconfiglib 1 fork that returns RST for the symbol, with links to other symbols for `select`, etc.

Kconfiglib 2 by default prints symbols using Kconfig syntax (with all the parent dependencies propagated). I think that format would work nicely in the documentation too. I could write a custom symbol printer that turns all the symbol names into links.