zephyrproject-rtos/zephyr #5626: Building samples failed

chsc, 2018-01-10:
Hello all, I have problems building the SDK sample applications. Specifically I was following the example at [STM32 Minimum Development Board](http://docs.zephyrproject.org/boards/arm/stm32_min_dev/doc/stm32_min_dev.html).

I get the following error after running `cmake -DBOARD=stm32_min_dev ..`:

```
CMake Deprecation Warning at /home/christoph/zephyr/cmake/app/boilerplate.cmake:38 (cmake_policy):
  The OLD behavior for policy CMP0000 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:1 (include)

-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.2", minimum required is "3.4") 
-- Selected BOARD stm32_min_dev
Zephyr version: 1.10.99
CMake Error at /home/christoph/zephyr/cmake/host-tools-zephyr.cmake:24 (file):
  file failed to open for reading (No such file or directory):

    /home/christoph/zephyr/sdk_version
Call Stack (most recent call first):
  /home/christoph/zephyr/cmake/host-tools.cmake:1 (include)
  /home/christoph/zephyr/cmake/app/boilerplate.cmake:211 (include)
  CMakeLists.txt:1 (include)

CMake Error at /home/christoph/zephyr/cmake/host-tools-zephyr.cmake:25 (if):
  if given arguments:

    "0.9.2" "VERSION_GREATER"

  Unknown arguments specified
Call Stack (most recent call first):
  /home/christoph/zephyr/cmake/host-tools.cmake:1 (include)
  /home/christoph/zephyr/cmake/app/boilerplate.cmake:211 (include)
  CMakeLists.txt:1 (include)

-- Configuring incomplete, errors occurred!
```

The sdk_version file seems to be missing, so I assume the build files are broken or outdated...

pfalcon, 2018-01-10:
Please make sure that it's not https://stackoverflow.com/questions/47081317/zephyr-cmake-error

SebastianBoe, 2018-01-11:
It's not.

This would happen if the SDK environment variable was pointing to the wrong directory.
It would also happen if the installed SDK version was so old it didn't have the sdk_version file.

Could you ls the SDK path environment variable?
Also, post the version of the SDK that is installed.

chsc, 2018-01-11:
Rechecked my installation. Seems that the SDK environment varaiable was pointing to a wrong folder.
_closed_

closed 2018-01-11 by chsc

aleandroconte, 2020-12-16:
Hi, I've a similar issue, what i see is this:
`gabriele@Gabriele-T490s:~/zephyr/zephyr-dwm1001/examples/ex_01a_simple_tx/build$ cmake -GNinja -DBOARD=nrf52_dwm1001 ..
Zephyr version: 1.14.0
-- Selected BOARD nrf52_dwm1001
CMake Error at /home/gabriele/zephyr/cmake/toolchain/zephyr/host-tools.cmake:44 (include):
  include could not find load file:

    /home/gabriele/zephyr/cmake/toolchain/zephyr/0.11.4/host-tools.cmake
Call Stack (most recent call first):
  /home/gabriele/zephyr/cmake/host-tools.cmake:1 (include)
  /home/gabriele/zephyr/cmake/app/boilerplate.cmake:379 (include)
  CMakeLists.txt:2 (include)

-- Found west: /home/gabriele/.local/bin/west (found suitable version "0.8.0", minimum required is "0.5.6")
validation.invalid
 --- All found errors ---
["Key 'settings' was not defined. Path: '/build'"]
Traceback (most recent call last):
  File "/home/gabriele/zephyr/scripts/yaml_to_cmake.py", line 68, in <module>
    main()
  File "/home/gabriele/zephyr/scripts/yaml_to_cmake.py", line 54, in main
    schema_data=yaml.safe_load(METADATA_SCHEMA)).validate()
  File "/home/gabriele/.local/lib/python3.6/site-packages/pykwalify/core.py", line 167, in validate
    error_msg=u'.\n - '.join(self.validation_errors)))
pykwalify.errors.SchemaError: <SchemaError: error code 2: Schema validation failed:
 - Key 'settings' was not defined. Path: '/build'.: Path: '/'>
CMake Error at /home/gabriele/zephyr/cmake/zephyr_module.cmake:59 (file):
  file STRINGS file
  "/home/gabriele/zephyr/zephyr-dwm1001/examples/ex_01a_simple_tx/build/zephyr_module_stm32.txt"
  cannot be read.
Call Stack (most recent call first):
  /home/gabriele/zephyr/cmake/app/boilerplate.cmake:394 (include)
  CMakeLists.txt:2 (include)

validation.invalid
 --- All found errors ---
["Cannot find required key 'build'. Path: ''", "Key 'samples' was not defined. Path: ''"]
Traceback (most recent call last):
  File "/home/gabriele/zephyr/scripts/yaml_to_cmake.py", line 68, in <module>
    main()
  File "/home/gabriele/zephyr/scripts/yaml_to_cmake.py", line 54, in main
    schema_data=yaml.safe_load(METADATA_SCHEMA)).validate()
  File "/home/gabriele/.local/lib/python3.6/site-packages/pykwalify/core.py", line 167, in validate
    error_msg=u'.\n - '.join(self.validation_errors)))
pykwalify.errors.SchemaError: <SchemaError: error code 2: Schema validation failed:
 - Cannot find required key 'build'. Path: ''.
 - Key 'samples' was not defined. Path: ''.: Path: '/'>
CMake Error at /home/gabriele/zephyr/cmake/zephyr_module.cmake:59 (file):
  file STRINGS file
  "/home/gabriele/zephyr/zephyr-dwm1001/examples/ex_01a_simple_tx/build/zephyr_module_mcuboot.txt"
  cannot be read.
Call Stack (most recent call first):
  /home/gabriele/zephyr/cmake/app/boilerplate.cmake:394 (include)
  CMakeLists.txt:2 (include)

validation.invalid
 --- All found errors ---
["Cannot find required key 'build'. Path: ''", "Key 'samples' was not defined. Path: ''"]
Traceback (most recent call last):
  File "/home/gabriele/zephyr/scripts/yaml_to_cmake.py", line 68, in <module>
    main()
  File "/home/gabriele/zephyr/scripts/yaml_to_cmake.py", line 54, in main
    schema_data=yaml.safe_load(METADATA_SCHEMA)).validate()
  File "/home/gabriele/.local/lib/python3.6/site-packages/pykwalify/core.py", line 167, in validate
    error_msg=u'.\n - '.join(self.validation_errors)))
pykwalify.errors.SchemaError: <SchemaError: error code 2: Schema validation failed:
 - Cannot find required key 'build'. Path: ''.
 - Key 'samples' was not defined. Path: ''.: Path: '/'>
CMake Error at /home/gabriele/zephyr/cmake/zephyr_module.cmake:59 (file):
  file STRINGS file
  "/home/gabriele/zephyr/zephyr-dwm1001/examples/ex_01a_simple_tx/build/zephyr_module_tfm-mcuboot.txt"
  cannot be read.
Call Stack (most recent call first):
  /home/gabriele/zephyr/cmake/app/boilerplate.cmake:394 (include)
  CMakeLists.txt:2 (include)

CMake Error at /home/gabriele/zephyr/cmake/toolchain/zephyr/generic.cmake:1 (include):
  include could not find load file:

    /home/gabriele/zephyr/cmake/toolchain/zephyr/0.11.4/generic.cmake
Call Stack (most recent call first):
  /home/gabriele/zephyr/cmake/generic_toolchain.cmake:55 (include)
  /home/gabriele/zephyr/cmake/app/boilerplate.cmake:395 (include)
  CMakeLists.txt:2 (include)

-- Loading /home/gabriele/zephyr/boards/arm/nrf52_dwm1001/nrf52_dwm1001.dts as base
-- Overlaying /home/gabriele/zephyr/dts/common/common.dts
CMake Error at /home/gabriele/zephyr/cmake/dts.cmake:93 (message):
  command failed with return code: No such file or directory
Call Stack (most recent call first):
  /home/gabriele/zephyr/cmake/app/boilerplate.cmake:396 (include)
  CMakeLists.txt:2 (include)

-- Configuring incomplete, errors occurred!
`

Is there any correlation with the upper error? @SebastianBoe 
thank you :)