zephyrproject-rtos/zephyr #5633: Optimize the CMake configuration time by reducing the time spent testing toolchain

SebastianBoe, 2018-01-11:
valgrind shows that a significant amount of time[0] is spent on TryCompile. TryCompile is invoked when we test the toolchain to determine what flags it supports.

Intuitively, the toolchain feature set is very static and should not impact the build system performance as much as it is. In addition, TryCompile is very verbose.

You would think that either maintaining some sort of database for the supported toolchains or caching
the toolchain results outside of the build directory would be a fairly straightforward optimization to do.

[0] 14% of the CMake execution time is spent in TryCompile.
"valgrind --tool=callgrind cmake -Boe -Hsamples/hello_world -DBOARD=qemu_cortex_m3"

SebastianBoe, 2018-02-09:
By set'ing

set(CMAKE_C_COMPILER ccache ${CMAKE_C_COMPILER})

I could observe that running CMake 

was reduced to 

real	0m0.862s

from

real	0m1.192s

which is a speedup of 40%, much more than what valgrind indicated.

This re-inforces that we should try to fix this performance issue. Perhaps using ccache (Unfortunately this does not help non-ccache users).

closed 2018-04-27 by carlescufi

marc-hb, 2019-09-18:
I just noticed while looking at strange cache bug #15551 that changing option parameters like for instance ${source_dir} in`-fmacro-prefix-map=${source_directory=..` regenerates brand new entries and a bit of noise in the cache. Suboptimal but I guess not big deal; just FYI.

PR #7102 (2018-04-27): Caching toolchain capabilities
This PR fixes #5633 and optimizes the CMake configure-time by 50%.

It does this by introducing a caching version of check_compiler_flag called zephyr_check_compiler_flag.

zephyr_check_compiler_flag documents itself like this:

> zephyr_check_compiler_flag is a part of Zephyr's toolchain
> infrastructure. It should be used when testing toolchain
> capabilities and it should normally be used in place of the
> functions:
> 
> check_compiler_flag
> check_c_compiler_flag
> check_cxx_compiler_flag
> 
> See check_compiler_flag() for API documentation as it has the sam [... cut]