zephyrproject-rtos/zephyr #5634: The dependency between the Kconfig source files and the Kconfig output is missing

SebastianBoe, 2018-01-11:
Editing prj.conf will correctly cause Kconfig to be re-executed when make is invoked. But editing Kconfig sources does not. The user must manually reconfigure (by calling cmake .. or make rebuild_cache).

### Solution proposal
Invoke

  set_property(DIRECTORY APPEND PROPERTY CMAKE_CONFIGURE_DEPENDS ${kconfig_source})

on all kconfig sources.

ulfalizer, 2018-05-09:
Simplest would be to always execute it I think.

If `sync_deps()` (see https://github.com/zephyrproject-rtos/zephyr/issues/4951) is called before each build and some `fixdep`-like scheme is added (see the `sync_deps()` docstring in Kconfiglib), then you could also automatically rebuild just the files that reference Kconfig symbols that have changed.

SebastianBoe, 2018-05-09:
The incremental build runtime is unacceptable if all incremental builds run CMake.

Do you see any issues with the solution proposal?

carlescufi, 2018-12-06:
@SebastianBoe @ulfalizer did we ever come to a conclusion on this issue? Are we using `sync_deps()` or @SebastianBoe's proposal?

ulfalizer, 2018-12-06:
Kconfiglib now makes a list of parsed Kconfig files available in `kconf.kconfig_filenames` at least. Might help, though it'd just be an optimization, as `sync_deps()` can detect all changes that matter.

See https://github.com/zephyrproject-rtos/zephyr/issues/4951 too.

PR #13587 (2019-03-07): kconfig: Have ninja re-run CMake when Kconfig sources change
Users often get confused when they change Kconfig sources and then
rebuild only to discover that nothing happens. To fix this we add a
dependency between re-running cmake, and all Kconfig sources, akin
to how touching CMakeLists.txt files cause CMake to re-run.

This fixes #5634
commit cabb35fc (2019-03-07): cmake: Kconfig: Minor refactorings in preparation for solving #5634

Minor refactoring of the build scripts in preparation for the patch
that resolves #5634.
commit f96c9bc7 (2019-03-07): kconfig: Have ninja Re-run CMake when Kconfig sources change

Users often get confused when they change Kconfig sources and then
rebuild only to discover that nothing happens. To fix this we add a
dependency between re-running cmake, and all Kconfig sources, similair
to how touching CMakeLists.txt files cause CMake to re-run.

This fixes #5634
closed 2019-03-07 by carlescufi