zephyrproject-rtos/zephyr #5645: build failures with asserts enabled/newlib: arch/arm/core/cortex_m/mpu/nxp_mpu.c

nashif, 2018-01-11:
```
[57/113] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/timer/cortex_m_systick.c.obj
FAILED: ccache /opt/sdk/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gcc -DCPU_MK64FN1M0VLL12 -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../../../../../../../kernel/include -I../../../../../../../arch/arm/include -I../../../../../../../arch/arm/soc/nxp_kinetis/k6x -I../../../../../../../arch/arm/soc/nxp_kinetis/k6x/include -I../../../../../../../arch/arm/soc/nxp_kinetis/include -I../../../../../../../boards/arm/frdm_k64f -I../../../../../../../include -I../../../../../../../include/drivers -Izephyr/include/generated -isystem /opt/sdk/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/lib/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/include -isystem /opt/sdk/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/lib/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/include-fixed -I/opt/sdk/zephyr-sdk-0.9.2/sysroots/armv5-zephyr-eabi/usr/include -I../../../../../../../ext/hal/cmsis/Include -I../../../../../../../ext/hal/nxp/mcux/devices/MK64F12 -I../../../../../../../ext/hal/nxp/mcux/components/phyksz8081 -I../../../../../../../ext/hal/nxp/mcux/drivers/. -I../../../../../../../subsys/net/lib/sockets/. -Os -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/sanity-out/frdm_k64f/samples/net/sockets/dumb_http_server/test/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main -std=c99 -Werror -DCONFIG_ASSERT=1 -D__ASSERT_ON=2 -Wno-deprecated-declarations -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-int -ffunction-sections -fdata-sections -mabi=aapcs --sysroot /opt/sdk/zephyr-sdk-0.9.2/sysroots/armv5-zephyr-eabi/usr -mthumb -mcpu=cortex-m4 -MD -MT zephyr/CMakeFiles/zephyr.dir/arch/arm/core/cortex_m/mpu/nxp_mpu.c.obj -MF zephyr/CMakeFiles/zephyr.dir/arch/arm/core/cortex_m/mpu/nxp_mpu.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/arch/arm/core/cortex_m/mpu/nxp_mpu.c.obj   -c /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/arm/core/cortex_m/mpu/nxp_mpu.c
In file included from ../../../../../../../include/kernel.h:24:0,
                 from ../../../../../../../include/device.h:11,
                 from /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/arm/core/cortex_m/mpu/nxp_mpu.c:7:
/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/arm/core/cortex_m/mpu/nxp_mpu.c: In function ‘_region_init’:
/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/arm/core/cortex_m/mpu/nxp_mpu.c:56:5: error: format ‘%x’ expects argument of type ‘unsigned int’, but argument 4 has type ‘uint32_t {aka volatile long unsigned int}’ [-Werror=format=]
     "Region %d base address got 0x%08x expected 0x%08x",
     ^
../../../../../../../include/misc/__assert.h:84:11: note: in definition of macro ‘__ASSERT’
    printk(fmt, ##__VA_ARGS__);                \
           ^~~
/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/arm/core/cortex_m/mpu/nxp_mpu.c:60:5: error: format ‘%x’ expects argument of type ‘unsigned int’, but argument 4 has type ‘uint32_t {aka volatile long unsigned int}’ [-Werror=format=]
     "Region %d end address got 0x%08x expected 0x%08x",
     ^
../../../../../../../include/misc/__assert.h:84:11: note: in definition of macro ‘__ASSERT’
    printk(fmt, ##__VA_ARGS__);                \
           ^~~
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
```

andrewboie, 2018-01-11:
so how do we actually fix this? if we use %lx won't that break non-newlib builds?

this is why stdint.h codes exist, however much some people dislike them...

galak, 2018-01-11:
To reproduce:

./scripts/sanitycheck  -T tests/misc/test_build/ -R -p frdm_k64f

galak, 2018-01-11:
@andrewboie casting to a (u32_t) I think will do the trick

MaureenHelm, 2018-01-11:
> @andrewboie casting to a (u32_t) I think will do the trick

It does. Preparing a patch

galak, 2018-01-11:
@MaureenHelm might have gotten there first.  Take a look at PR #5647

MaureenHelm, 2018-01-11:
> @MaureenHelm might have gotten there first. Take a look at PR #5647

Ok, thanks

PR #5647 (2018-01-11): arm: nxp_mpu: Fix build issue when asserts enabled
When asserts are enabled we run into an issue with newlib and types of
printf style formatters not matching.  The easy fix to this is to cast
the uint32_t to u32_t to make things consistent with or without newlib
enabled.

This fixes #5645
commit cd7ca953 (2018-01-11): arm: nxp_mpu: Fix build issue when asserts enabled

When asserts are enabled we run into an issue with newlib and types of
printf style formatters not matching.  The easy fix to this is to cast
the uint32_t to u32_t to make things consistent with or without newlib
enabled.

This fixes #5645
closed 2018-01-11 by MaureenHelm