zephyrproject-rtos/zephyr #5650: i2c driver test on ESP32 fails with error

nniranjhana, 2018-01-12:
Using GPIO 25 as SCL and GPIO 26 as SDA (tried with other GPIO pins from default ones too) for the i2c module for ESP32, with a sensor GY271 fails with error 'FAILED TO CONFIGURE SENSOR'. However, passes with GPIO bit banging with ESP32. Hence, driver needs debugging.

Attached log below

nniranjhana, 2018-01-12:
[esp32_i2c+not+working.txt](https://github.com/zephyrproject-rtos/zephyr/files/1625107/esp32_i2c.not.working.txt)

kumarvikash1, 2018-01-12:
Logs snapshot : 

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00                                               
mode:DIO, clock div:2                                                                                                 
load:0x3ffb0000,len:96                                                                                                
load:0x3ffb0060,len:704                                                                                               
load:0x3ffb0320,len:2336
load:0x40080000,len:1024
load:0x40080400,len:96
load:0x40080460,len:11216
load:0x400c0000,len:0
entry 0x4008069c
Running test suite i2c_test
===================================================================
starting test - test_i2c_gy271
Fail to configure sensor GY271

    Assertion failed at /home/nj/zephyr/zephyr/tests/drivers/i2c/i2c_api/src/test_i2c.c:128: test_i2c_gy271: test_gy271() == TC_PASS is false
FAIL - test_i2c_gy271.
===================================================================
starting test - test_i2c_burst_gy271
Fail to write to sensor GY271

    Assertion failed at /home/nj/zephyr/zephyr/tests/drivers/i2c/i2c_api/src/test_i2c.c:133: test_i2c_burst_gy271: test_burst_gy271() == TC_PASS is false
FAIL - test_i2c_burst_gy271.
===================================================================
===================================================================
PROJECT EXECUTION FAILED

lpereira, 2018-01-16:
There are two patches addressing I2C on ESP32: #5690 and #5694.

iha, 2018-01-16:
FYI, I'm reviewing the I2C driver on ESP32.

carlescufi, 2018-02-06:
@lpereira @iha Is this still relevant?

iha, 2018-02-06:
Hi @carlescufi , I made these two PRs to fix the I2C Master: https://github.com/zephyrproject-rtos/zephyr/pull/5850 and https://github.com/zephyrproject-rtos/zephyr/pull/5851 . The https://github.com/zephyrproject-rtos/zephyr/pull/5850 needs some change but with these two PRs the I2C driver works.

I tested with a LCD (HD44780) and an accelerometer (MMA7660).

PR #5850 (2018-02-07): drivers: pinmux: esp32: Fix using gpio as IO at the same time.
Since PINMUX_INPUT_ENABLED == 1 and PINMUX_OUTPUT_ENABLED == 0,
we can not set a gpio port as input and output at the same time,

So we always set the gpio as input. Thus, the gpio can be used on
I2C drivers for example.

Fixes #5650
closed 2018-02-07 by nashif