zephyrproject-rtos/zephyr #5653: STM32 boards: Generic guidelines for pin configuration

erwango, 2018-01-12:
Following discussions on #3483 (fixed by #5515), generic guidelines for STM32 based boards pin configuration have been defined.

For STM32 based boards ported on Zephyr, it is recommended to define and provide pin configuration based on the following elements:
- Provide pin configuration to enable use of HW available on board like button or LED and communication interfaces (UART, I2C, SPI) to connect various peripherals such as sensors, embedded modules (BLE, Wi-Fi) or Virtucal Com Port (as provided by STLink), external communication interfaces (USB, Ethernet) or embedded memories (Flash, ...)... 
- Provide pin configuration matching silkscreen notations when available, in particular when it enables compatibility with expansion boards (such as Arduino connectors allowing use of Adruino compatible shields).

Aim of these guidelines is to provide coherency among boards configuration, ease board porting and reviews and providing assumptions for easier tests implementation.

erwango, 2018-01-12:
@ydamigos , @superna9999 , @mbolivar : As discussed

superna9999, 2018-01-12:
Ok for me

tagunil, 2018-03-13:
Could you explain me to which aspects of board porting these rules should be applied? Pinmux, board dts, Kconfig.defconfig, board defconfig? Which SPI/I2C ports I should add to a new Nucleo board device tree?

erwango, 2018-03-14:
@tagunil : Initially it was supposed ot be applied to pinmux. But it is right this extends to generic default board configuration. I'll see next week if extending this to more generic directives make sense.

erwango, 2018-05-22:
Could be close following merge of #6947

closed 2018-05-22 by erwango