zephyrproject-rtos/zephyr #5660: doc: make: make htmldocs fails in genrest.py/kconfiglib.py

b0661, 2018-01-12:
make htmldocs stops execution with the following error message. 

I´m on zephyr/master commit 15ca6da4c8a3130650de70061f92c9bb9925cb4d.

>
(cat zephyr.doxyfile ; echo "STRIP_FROM_PATH=/media/bernhard/zephyr/zephyr-bobby2" ) | doxygen -  2>&1 | tee doc.log
scripts/extract_content.py
srctree=../ KERNELVERSION=1.9.99 SRCARCH=x86 python3 scripts/genrest/genrest.py ../Kconfig reference/kconfig/
../Kconfig.zephyr:15: warning: The symbol ENV_VAR_SYM_BOARD_DIR references the non-existent environment variable ENV_VAR_BOARD_DIR and
 will get the empty string as its value. If you're using Kconfiglib via 'make (i)scriptconfig', it should have set up the environment 
correctly for you. If you still got this message, that might be an error, and you should email ulfalizer a.t Google's email service.
../Kconfig.zephyr:19: warning: The symbol ENV_VAR_SYM_ARCH references the non-existent environment variable ENV_VAR_ARCH and will get 
the empty string as its value. If you're using Kconfiglib via 'make (i)scriptconfig', it should have set up the environment correctly 
for you. If you still got this message, that might be an error, and you should email ulfalizer a.t Google's email service.
Traceback (most recent call last):
  File "scripts/genrest/genrest.py", line 100, in <module>
    conf = kconfiglib.Config(sys.argv[1])
  File "/media/bernhard/zephyr/zephyr-bobby2/doc/scripts/genrest/kconfiglib.py", line 219, in __init__
    self.top_block = self._parse_file(filename, None, None, None)
  File "/media/bernhard/zephyr/zephyr-bobby2/doc/scripts/genrest/kconfiglib.py", line 606, in _parse_file
    visible_if_deps, res)
  File "/media/bernhard/zephyr/zephyr-bobby2/doc/scripts/genrest/kconfiglib.py", line 696, in _parse_block
    self._parse_file(s, parent, deps, visible_if_deps, block)
  File "/media/bernhard/zephyr/zephyr-bobby2/doc/scripts/genrest/kconfiglib.py", line 606, in _parse_file
    visible_if_deps, res)
  File "/media/bernhard/zephyr/zephyr-bobby2/doc/scripts/genrest/kconfiglib.py", line 696, in _parse_block
    self._parse_file(s, parent, deps, visible_if_deps, block)
....
  File "/media/bernhard/zephyr/zephyr-bobby2/doc/scripts/genrest/kconfiglib.py", line 696, in _parse_block
    self._parse_file(s, parent, deps, visible_if_deps, block)
  File "/media/bernhard/zephyr/zephyr-bobby2/doc/scripts/genrest/kconfiglib.py", line 605, in _parse_file
    return self._parse_block(_FileFeed(filename), None, parent, deps,
  File "/media/bernhard/zephyr/zephyr-bobby2/doc/scripts/genrest/kconfiglib.py", line 3291, in __init__
    self.lines = f.readlines()
RecursionError: maximum recursion depth exceeded while calling a Python object
Makefile:76: die Regel für Ziel „kconfig“ scheiterte
make: *** [kconfig] Fehler 1

Cheers
Bobby

nashif, 2018-01-12:
this is already fixed.

b0661, 2018-01-13:
Tested with newest master - works now. Thank you

closed 2018-01-13 by b0661

ulfalizer, 2018-01-17:
I just added recursive `source` to detection to Kconfiglib, so similar problems should be a bit easier to debug now:

```
Kconfig.zephyr:11: warning: 'option env="KERNELVERSION"' on symbol KERNELVERSION has no effect, because the environment variable KERNELVERSION is not set
Kconfig.zephyr:15: warning: 'option env="ENV_VAR_BOARD_DIR"' on symbol ENV_VAR_SYM_BOARD_DIR has no effect, because the environment variable ENV_VAR_BOARD_DIR is not set
Kconfig.zephyr:19: warning: 'option env="ENV_VAR_ARCH"' on symbol ENV_VAR_SYM_ARCH has no effect, because the environment variable ENV_VAR_ARCH is not set
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "kconfiglib.py", line 641, in __init__
    self.top_node)  # prev_node
  File "kconfiglib.py", line 1552, in _parse_block
    prev_node)
  File "kconfiglib.py", line 1552, in _parse_block
    prev_node)
  File "kconfiglib.py", line 1552, in _parse_block
    prev_node)
  File "kconfiglib.py", line 1552, in _parse_block
    prev_node)
  File "kconfiglib.py", line 1548, in _parse_block
    self._enter_file(self._expand_syms(self._next_token()))
  File "kconfiglib.py", line 1126, in _enter_file
    in reversed(self._filestack))))
kconfiglib.KconfigSyntaxError: 
arch//Kconfig:301: Recursive 'source' of 'arch//Kconfig' detected. Check that environment variables are set correctly.
Backtrace:
arch//Kconfig:301
arch/Kconfig:301
Kconfig.zephyr:21
Kconfig:10
```

Looks like this script is still using Kconfiglib 1 though.