zephyrproject-rtos/zephyr #5691: FCB: add entry-value ID and related features.

nvlsianpu, 2018-01-16:
Add entry ID and related features to FCB.
This ID is  most simple substitute  for a file name.
Idea is that it will be  selectable using Kconfig.

Related functionalities:
- searching entry of certain ID
- overwrite capability so by writing a new entry-value with the same entry number the previous is considered overwritten.
- When rotation is required entries that have not been written again are copied to the front of the buffer.
This allows one flash buffer to be used for storage of (nearly) constant data and frequently changing data.

@Laczen proposed the module [simple-FCB](https://github.com/Laczen/zephyr/tree/master/subsys/sfcb) which has expected functionalities.

nvlsianpu, 2018-01-16:
@Laczen Can you comment on this issue? Is it possible that you will contribute such features to the zephyr's FCB?

pfalcon, 2018-01-16:
What is FCB?

Laczen, 2018-01-17:
Some comment on this issue:

I created the simple fcb library to allow storage in program flash of configuration variables, more specific for btmesh. btmesh is a somewhat specific case because it needs storage of both variables that can be regarded as constant and variables that change frequently. I was looking for a solution that would allow both types of variables to be stored in one (configurable) area of flash. I developed a solution that stores variables as an ID-value pair. The solution allows a quite large amount (e.g 65536 - 2) different variables to be stored in flash. The library uses flash as a circular buffer overwriting old data when space is insufficient. However before it overwrites the data the library checks if it needs to copy old data to the front of the buffer, and does so if required.

A typical use case would be storing data for a wireless light switch where you would like to store the address of the switch and also the state of the switch in case of electricity outages. The address will be constant, but the state changes every time the light is switched. Both can be stored in the same flash area. If the electricity comes back on the light switch can determine its address and the last state so that it can be reset to its last state.

There are many other use cases that can be determined, e.g. different loggings were the ID value is used to determine what is logged, ...

The library makes sure that for all ID-value pairs that are written in flash it wil keep at least one in flash for each ID.

The library has a similar API as fcb from mynewt. As I wrote it for a 16Kb ram device it needs only 44 bytes of ram (including a mutex).

I would happily contribute the library to zephyr. I have only one concern: zephyr fcb is build on top of flash_map, and flash map requires quite some ram, that is not available on 16Kb devices.

Kind regards,

Jehudi

nvlsianpu, 2018-09-17:
This request was already resolved/satisfied by NVS introduction which.

closed 2018-09-17 by nvlsianpu