zephyrproject-rtos/zephyr #5697: Driver API review/cleanup/rework

galak, 2018-01-16:
This issue will be used to track all driver subsystem rework/cleanups:
- [ ] adc
- [ ] aio
- [ ] bluetooth
- [ ] clock_control
- [ ] console
- [ ] counter
- [ ] crypto
- [ ] display
- [ ] dma
- [ ] entropy
- [ ] ethernet
- [ ] flash
- [ ] gpio
- [ ] grove
- [ ] i2c
- [ ] i2s
- [ ] ieee802154
- [ ] interrupt_controller
- [ ] ipm
- [ ] led_strip
- [ ] net
- [ ] pci
- [ ] pinmux
- [ ] pwm
- [ ] rtc
- [ ] sensor
- [ ] serial
- [ ] shared_irq
- [ ] spi
- [ ] timer
- [ ] usb
- [ ] watchdog

b0661, 2018-01-31:
@galak can pinctrl PR #5799 also be added  here?

carlescufi, 2018-01-31:
@b0661 done

Mani-Sadhasivam, 2018-07-18:
@galak @carlescufi Can we add LED API to this issue?

carlescufi, 2018-07-18:
@Mani-Sadhasivam done, is that what you meant?

Mani-Sadhasivam, 2018-07-18:
@carlescufi No, I mean to add LED entry to the list of API's here. I will work on extending the API further.

Mani-Sadhasivam, 2018-07-18:
@carlescufi Moreover, issue #8946 belongs to LED API and not LED Strip.

vaussard, 2018-08-17:
Issues related to the PWM API: #6958 and #9507

galak, 2019-04-16:
@mbolivar to move this PR into a project.

nvlsianpu, 2019-04-29:
@galah @carlescufi  flash driver documentation issue: [flash: should write_protection be emulated?](https://github.com/zephyrproject-rtos/zephyr/issues/15729)

mbolivar, 2019-04-30:
> @mbolivar to move this issue into a project.

Finally got around to this: https://github.com/zephyrproject-rtos/zephyr/projects/18

Is that pretty much what people are expecting? Should we close this if so?

andrewboie, 2019-04-30:
I think we need clarity on what is going on with the new async serial API, and the old CONFIG_UART_INTERRUPT_DRIVEN API, as seen here: https://github.com/zephyrproject-rtos/zephyr/pull/14854

henrikbrixandersen, 2022-11-02:
@galak Does this issue still provide value or can it be closed?

galak, 2022-11-02:
> @galak Does this issue still provide value or can it be closed?

going to close it.  We can re-open if needed.

closed 2022-11-02 by galak