zephyrproject-rtos/zephyr #5699: Zephyr installs a broken pyOCD

mbolivar, 2018-01-16:
The Zephyr file `scripts/requirements.txt` pulls in pyOCD:

https://github.com/zephyrproject-rtos/zephyr/blob/master/scripts/requirements.txt

That file is used to install Python 3 dependencies using pip(3) in Zephyr's getting started documentation:

http://docs.zephyrproject.org/getting_started/installation_linux.html#installing-requirements-and-dependencies
http://docs.zephyrproject.org/getting_started/installation_mac.html#installing-requirements-and-dependencies
http://docs.zephyrproject.org/getting_started/installation_win.html#installing-requirements-and-dependencies

pyOCD doesn't work on Python 3, though, as described in this issue:

https://github.com/mbedmicro/pyOCD/issues/208

There are two changes that are needed to fix pyOCD use on Zephyr until that issue is resolved.

1. Zephyr must remove the pyOCD from scripts/requirements.txt
2. The Zephyr documentation must clarify that if pyOCD is being used to flash a board (any board), it must be installed with Python 2

A fix for this issue requires checking around all the individual board documentation pages in addition to the getting started / installation instructions.

dbkinder, 2018-01-16:
Note that the Zephyr installation instructions don't even install Python 2 or pip(2), so do we need to install them too?

mbolivar, 2018-01-18:
Users which require pyOCD need Python 2 and to install it with Python 2. That's wasteful and probably confusing for users which don't require pyOCD.

PR #6003 (2018-02-07): doc: install pyocd using pip2
pyocd does not exist as a python 3 module, use pip2 to install it.

Fixes #5699
closed 2018-02-07 by nashif