zephyrproject-rtos/zephyr #5703: Provide NFC Library

erwango, 2018-01-17:
NFC is a important part of IoT ecosystem and now part of some basic use cases that should be supported by Zephyr.
It is also available on some boards and ICs supported by Zephyr.
A NFC library ported onto Zephyr would allow to extend Zephyr offer and provide more complete use cases.

First priority would be NFC Tag functionality would be sufficient. NFC reader capability would be a lesser prio considering targeted use case by Zephyr devices.

First step of the study would be to establish requested services level (from basic memory write to full NDEF support).

rodrigopex, 2018-08-07:
@nashif any progress on that? Can I start to work to create the API/Driver?

nullr0ute, 2018-08-07:
@rodrigopex I would suggest just get going on it if you have the cycles

rodrigopex, 2018-08-07:
What do you mean with cycles?

nullr0ute, 2018-08-07:
If you have the time to work on it (the cycles) I'd probably just do so, maybe check on IRC

grochu, 2018-08-28:
> Guys, any progress on that? We will need NFC for a project. I am available to help on that if you can support me (information and help) on creating it. The first I will focus initially on the driver for nRF52832 device.

We are working on adding an NFC driver supporting nRF52832 and nRF52840 chips to the NRFX driver package (and consequently to Zephyr). The timeline is around end of September - beginning of October.

rodrigopex, 2018-08-28:
Great! If you need help to test it when ready, let me know.

Best regards,
Rodrigo Peixoto

erwango, 2018-08-28:
> We are working on adding an NFC driver supporting nRF52832 and nRF52840 chips to the NRFX driver package (and consequently to Zephyr).

Might be worth sharing the API once you have something ready. The sooner the API is discussed the better IMHO.

grochu, 2018-08-28:
> Might be worth sharing the API once you have something ready. The sooner the API is discussed the better IMHO.

Agree, we can start discussing the API once it gets stable, but let me clarify 2 things already:

1. What we are going to provide in the first step is a HW driver for the nRF52 chips, so roughly it will cover what is supported by the NFC peripheral in these MCUs. We are not attempting to propose a general NFC driver API for Zephyr in this step.
2. To reveal some details on the driver API, I can say it is rather low-layer and corresponds to chapter 6 of the "NFC Forum Digital Protocol Technical Specification 2.0" (protocol layer equivalent to ISO 14443-3: _Initialization and anti-collision_).

aaronhshi, 2018-08-29:
@grochu is proposing/developing a general NFC driver API for Zephyr in your guy's plan for the future? This issue is something I'm very interested in and was hoping to develop and be apart of in providing a general NFC driver API for Zephyr!

grochu, 2018-08-29:
@aaronhshi yes, we are planning to propose a general NFC API for Zephyr as well. We will come up with an API-only pull request to start the discussion.

thomasthep-depone, 2019-08-20:
Any updates regarding the status of NFC?

There's a pull-request https://github.com/zephyrproject-rtos/zephyr/pull/10117 which has since last been touched on April the 30th. It's been marked stale since.

kshitizdange, 2020-10-14:
Any update on this? 
I do see that the `nfc_hello` sample that existed in `2.3.0` release, has been removed from `2.4.0` release.

selman-nus, 2021-02-05:
> We are working on adding an NFC driver supporting nRF52832 and nRF52840 chips to the NRFX driver package (and consequently to Zephyr). The timeline is around end of September - beginning of October.

@grochu Is there any snippet yet?

grochu, 2021-02-05:
There is an NFCT driver in the hal/nordic repo: _modules/hal/nordic/nrfx/drivers/src/nrfx_nfct.c_ which supports nRF52 and nRF53 families (devices which have NFCT). At the moment there are no plans of adding a generic Zephyr NFC driver or upper-layer libraries, but we are open to support such activity.

adamfowleruk, 2021-04-02:
Happy to see the Nordic nRF52 and 53 supported. Wanted to flag that this may be needed by the Herald Project for Digital Contact Tracing wearable checking-in, or out of band pairing for bluetooth smartphone with wearable (if possible... not sure if any SOCs have readers...), or by a number of covid credentials projects. Our project uses nRF52 and 53 chips as our reference hardware platform so with the comment from @grochu I think we are fine for now.

ck-telecom, 2021-09-26:
any update for generic NFC library? could we steal from MBed?

matt-wood-ct, 2023-03-09:
Is anyone working on this, I could really use it

marcel-ithinx, 2023-09-20:
I'm working on a driver for the NXP CLRC661 and CLRC663 and ISO15693 and NXP slix protocol layers. All of that is still in work or internal review. I'll try to open a PR around mit october.

pdgendt, 2023-09-20:
Hi @marcel-ithinx, I've also worked on an NFC proposal (which is far from complete), but maybe can help out for more ideas:
https://github.com/zephyrproject-rtos/zephyr/compare/main...pdgendt:zephyr:nfc-subsys

I've added the NXP PN5180 driver and ISO14443A subsystem, and tested both reader mode and HCE (Host Card Emulation).

You (or anyone) can ping me if desired :)

MarcelKr, 2023-09-20:
(switching to proper account, oops)

@pdgendt Uh, that looks great! I didn't implement the NFC driver API abstraction (nfc.h) yet (which you already did). I'm on vacation the next two weeks but after that will have a closer look and probably will try to adapt my stuff to use your nfc.h. Thanks for your response :)

HesselM, 2023-10-17:
Hi all! 

Last year I've been working with NFC on both nRF and NXP platforms. Part of our zephyr port to the qn9090 I actually made a NFC (tag) subsystem, supporting the nRF-NFCT drivers and the nth3x-nfc tag (which is internally located within the qn9090 die). 

I never bothered making it public, but as part of the qn9090 zephyr 3.4 port, i moved it to public space.
You can find my approach at : https://github.com/Sendrato/zephyr/tree/v3.4-branch-nfc.

The approach is 90% based on the approach nordic took. 
The nordic-drivers are probably outdated as they are from June 2021 or so?

HesselM, 2023-10-19:
Hi all, 

An update: I've patches bugs, cleaned the branch/commit and upgraded all nRF drivers. 
The proposed NFC subsystem can be checked at https://github.com/Sendrato/zephyr/tree/v3.4-branch-nfc

I've included a sample which should run on every nordic device with NFC and devices using the nt3h2 driver.
A board-configuration-overlay for the dk6_qn9090 and mg100 are included.

Hessel.

matt-wood-ct, 2024-02-19:
This is still interesting, any movement on this?

gregistech, 2024-08-13:
> This is still interesting, any movement on this?

matt-wood-ct, 2025-01-06:
I do still feel this is essentially a necessary addition, why can't we just adopt Nordics NFC implementation they wrote for supporting the NFC-A peripherals on their chips, I understand the current files have a Nordic license on, are they actually opposed to allowing its use on non Nordic platforms? Is there a wider issue at play?

razvanphp, 2025-08-07:
@HesselM do you think we could work together to add PN7160 support via NCI 2.0 as well? Seem from the supported products list that the chips should work with a common driver: https://www.nxp.com/applications/technologies/security/industrial-security/nfc-reader-library-software-support-for-nfc-frontend-solutions:NFC-READER-LIBRARY

We would really need it in our project.

ShahinHaque97, 2026-01-21:
Has there been any movement on this?