zephyrproject-rtos/zephyr #5707: Move fixups definitions in .fixup files under dts/

erwango, 2018-01-17:
extract_dts_includes.dts script allows to gather and merge .fixup files accross the tree.

This allows to factorize most of the definitions in boards/*/*/dts.fixup in new dts/*/*.fixup files.
This way, one would not need to bother with dts.fixup when porting new board

DWagenk, 2018-01-26:
@erwango where do the new dts.fixup files go? 
for example moving SPI fixup from boards/arm/nucleo_f091rc to ? 
my guess would be a dts.fixup file in arch/arm/soc/st_stm32/stm32f0/dts.fixup, is this correct? 
And are there changes needed in the build system somewhere?

erwango, 2018-01-26:
Yes, this is under arch, you're right (editing the issue header).
There is nothing you should do to get it working, possibility has been enabled since some time now in the device tree parsing script.
I haven't tested it yet, but it seems galak started to move things in #5828.

galak, 2018-01-26:
I added support for multiple -f options to the script in commit aed577a853a24c49ad63464a571d5610063526b8.  I added recent cmake support to b452e1bbbb275dc1ac16084e82b7ca7690690564 to support an SoC series level fixup.

erwango, 2018-01-29:
fixed by #5859

DWagenk, 2018-01-29:
@erwango does this only effect STM32 (wich is fixed now), or shouldn't it stay open for other SOC Families?

erwango, 2018-01-29:
@DWagenk : You're right, this was a generic ticket, let me re-open it.

galak, 2018-02-21:
Moved NXP in PR #6297

alext-mkrs, 2018-06-16:
Looking for simple initial stuff (and to get a hang of DTS use in Zephyr) I've come across this one and looks like at least none of the ARC and x86 boards have `dts.fixup` files - shouldn't those be marked as done in the issue checklist?

Also, @galak mentioned that NXP is done too in #6297 (just a comment above this one) - should it be ticked off as well?

Finally, here's the current list of fixup files under `boards/` - I've checked several ones randomly (96b_argonkey, nrf52_pca20020, frdm_k64f) and it seems to me that none of them lend themselves to this "upleveling", as they have real board-specific info. Is that a correct understanding? If not, any pointers as to what else needs to be done here? I'm not that familiar with DTS yet, but it's namely my goal here - learn and help the project at the same time, so I'd be glad to dig into details.

```
boards/arm/frdm_k64f/dts.fixup
boards/arm/olimexino_stm32/dts.fixup
boards/arm/mps2_an385/dts.fixup
boards/arm/frdm_kw41z/dts.fixup
boards/arm/96b_neonkey/dts.fixup
boards/arm/disco_l475_iot1/dts.fixup
boards/arm/96b_argonkey/dts.fixup
boards/arm/hexiwear_k64/dts.fixup
boards/arm/nrf52_pca20020/dts.fixup
boards/arm/96b_carbon/dts.fixup
```

galak, 2018-09-12:
@erwango can you review this.

erwango, 2018-11-22:
Only MPS board remains.
Though, since there is no other board using this SoC, there is no code duplication to factorize there.
Let's close ticket right now

closed 2018-11-22 by erwango