zephyrproject-rtos/zephyr #5714: 15.4 features required for OpenThread certification

KaSroka, 2018-01-17:
List of features needed to be implemented for OpenThread in Zepyr certification. 
1. Energy scan
2. Src address matching (filtering of incoming frames basing on ext/short src address)

Not needed for certification but also missing
1. Prmoiscuous mode

tbursztyka, 2018-01-18:
For promiscuous mode, it's up to the driver to support it. see include/net/ieee802154_radio.h : enum ieee802154_hw_caps. 

About "energy scan", isn't rssi value on net_pkt sufficient? (you get the lqi as well)

For src address matching, I guess it will require to change and expend the enum ieee802154_filter_type to differentiate local_ieee/short_addr and src_ieee/short_addr and then again: it will up to the drivers to support it, if they are able to.

rveerama1, 2018-01-18:
Look at this 
https://github.com/zephyrproject-rtos/zephyr/pull/5202/commits/85f5e0cfb61a1462f4bf012aa2434013efb9cb22#diff-e281efeece521183d53e9fe70d52aed6

A PR submitted for virtual RPL mesh hub over qemu. and Idea of 
IEEE802154_UPIPE_PROMISCUOUS mode.

hubertmis, 2018-01-18:
@tbursztyka , Energy scan (passive scanning) is a procedure when radio is measuring any energy on given channel. It does not receive any packets during this procedure. It just checks if there is any activity (possibly other protocols like WiFi) on given frequency. Because of that the driver cannot receive packets and check RSSI instead.

tbursztyka, 2018-01-18:
@rveerama1  which has nothing to do with proper 15.4 radio chip promiscuous mode. What the guy implemented is, in fact, dst address matching (which is the current address filtering 15.4 radio api exposes). I'll review the patch

@hubertmis That I know. thus my question: "energy scan" (rssi evaluation) is done also at reception (with slight differences on how devices compute it however).

So the stand-alone one is necessary as well then?

For the record, I am asking this because some old requirement requested to get the lqi separately, and that was super important and what not, and get_lqi was then introduced to become, finally... completely useless.

hubertmis, 2018-01-18:
@tbursztyka I can't image how could I replace energy scan with simple rx procedure. Let's say I have a few WiFi networks and I want to start new 802.15.4 network. I'm enabling 802.15.4 device that performs energy scan. If the device entered rx mode instead of energy scan, it would not receive any 802.15.4 frames and it would not know what is max energy level on each of scanned channels (RSSI and LQI are reported for received frames, driver does not report these values if there are no incoming frames).

tbursztyka, 2018-01-18:
@hubertmis so OT _does_ need to grab the RSSI out of any frame reception?

A yes/no answer will be sufficient.

hubertmis, 2018-01-18:
@tbursztyka The answer is not so easy :)
No, in general. I think there is an optional jamming detection procedure that checks RSSI in RX mode when no frame is being received.
But Energy Detection procedure (as specified in 802.15.4) is required.

tbursztyka, 2018-01-18:
ook

Energy Detection is the ability for the phy device to provide the RSSI at any time. So that's covered by the device.

Then what about a get_rssi() formalized as:
- if device is transmitting, it will return -EBUSY
- if device is effectively receiving (a frame is being handled and pushed into the device's FIFO), it will return -EBUSY (or, it could wait until reception has ended and return the same RSSI value that was computed at frame reception, either way)
- if device is in rx mode but no frame is being handled, it will do what's necessary to grab the RSSI and return it. (some device can do it on the fly, some needs to restart to a specific rx mode first, ...)

Or, and that one means a bit more work, but is less ambiguity-prone I think: 
- we introduce a rx() function so the L2 can dictate when the device has to go to receive mode (same as tx() for transmitting)
- get_rssi() would work if only the device is not in rx/tx (note that grabbing the rssi will temporary switch the device to receiving mode, but - usually afaik - without frame handling logic). Otherwise it will return -EBUSY.

(unless introducing such rx() is an issue for OT integration, I don't know).

hubertmis, 2018-01-19:
Also in OT we need to add parameter to define time of energy scan procedure:
`otError otPlatRadioEnergyScan(otInstance *aInstance, uint8_t aScanChannel, uint16_t aScanDuration);`
How do you suggest to use `get_rssi()` and perform 5 ms of scanning? Call this function in a loop? It would be inefficient if hardware supports energy detection procedure with configurable time.

tbursztyka, 2018-01-19:
That's the higher level functionality you are talking about. The driver api provide the way to switch channel, and (will provide) the way to get the rssi.

That will permit @KaSroka to implement the function you are talking about.

Now that you mention timing, on driver api levele maybe instead of get_rssi() it should be energy_detect(<duration>). On device level it's the same as going to rx, but without the frame handling logic enabled. It would behave the same way as previously described for get_rssi, plus the time handling.

KaSroka, 2018-01-19:
So you want to create `energy_detect()` which will accept time as parameter and return detected energy? And it will be in 15.4 driver API, so every vendor could implement it depending on their radio capabilities (calling rx in a loop or something like that or as we have - special API call for this functionality).

tbursztyka, 2018-01-19:
@KaSroka Implementing this will all depends on chip's logic yes. In cc2520 case, it just needs to go on radio rx mode, without frame handling. But it does not manage any delay itself, so that part would be in driver side. It would return the RSSI, but as a int because we need to be ready to return errors as well (-EBUSY, ...)

KaSroka, 2018-01-19:
What is the convention for RSSI in Zpehyr? Is it in dBm or some other value?

jfischer-phytec-iot, 2018-01-19:
>On device level it's the same as going to rx, but without the frame handling logic enabled. It would behave the same way as previously described for get_rssi, plus the time handling.

Energy detect is a CCA mode. Would not it be better to rework `ieee802154_radio_api.cca`? Making it in rx does not seem to me to be a decent solution. For the MCR20A and KW41Z driver, ED Final Average Value can be read out after the CCA sequence.

KaSroka, 2018-01-19:
OT expects values in dBm

jfischer-phytec-iot, 2018-01-19:
>it and it gives you a boolean, plus you cannot ask for a longer delay (that I guess depends on chip's logic again).

Yes, it depends on transceiver logic, for the MCR20A and KW41Z transceivers, ED Final Average Value can be read out after the CCA sequence and there is a `Continuous CCA mode` beside `Standalone CCA`.

tbursztyka, 2018-01-19:
@jfischer-phytec-iot ok so refining the cca() call is not the solution. However, in your case you would implement the ed() through this continuous cca mode I guess.

@KaSroka dbm should be fine as a result, so ed() would not return the rssi. Now depends on the chip's logic: in case of cc2520 the dbm value has to be computed from the given rssi. (calculus being chip dependent here)

jfischer-phytec-iot, 2018-01-19:
@tbursztyka Also doing it in the rx is very cc2520 specific :smile: (from the cc2520 datasheet: To completely avoid receiving frames during energy detection scanning, set FRMCTRL0.RX_MODE = 0b11 and then (re)start RX.)

Generally what is behind it, from the Thread specification (I have currently only v1.1.0) the requirements is that _device must perform IEEE 802.15.4 ED Scans_, and the parameters for MGMT_ED_SCAN are: channel mask, count, period(milliseconds), scan duration(milliseconds).

KaSroka, 2018-02-02:
There is one more thing to do before we can certify: https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/openthread/platform/radio.c#L110

tbursztyka, 2018-03-15:
@KaSroka so you need the full ack frame + fcs? 
Is OT really checking the FCS? Because the "spoofed" version you showed, is made of 0s for the FCS.
Just asking because in driver side this could have effects if the true FCS is required.

dleach02, 2018-03-16:
I believe it can be enabled to check. At a minimum, it expects the receive packet to be passed up to include the FCS which is why the Nordic and NXP enablement had to add the little quirk in the RX code. I've thought a bit about what we are having to do in the 15.4 drivers to support a different L2 and this is one area I would normalize our stack to receive the FCS as well. My thinking is that I would like to see the drivers have no conditional code for differing L2s since it should have no knowledge of what is attaching to it.

dleach02, 2018-03-16:
With respect to the energy scan, I looked at the OT code and the KW41Z driver and it is having the driver handle the duration of the scan. In the case of the KW41Z, there is a HW phy timer resource that is set to the timeout while the radio is put into energy detect mode. Then when the scan is done, the driver lets the OT know and OT then calls to get the results (I think).

tbursztyka, 2018-03-16:
@dleach02 The issue with including true FCS or not on driver side is about HW FCS verification, which is really something you want, there is no point verifying the FCS by ourselves: too slow, too high in the stack when this could be already done at the bottom. TI cc2520 is annoying in that regard as it verifies the FCS (if enabled to do so) and modifies the content there to tell if it passed or not. I don't know about other devices, maybe they are smarter. Keeping the FCS into net_pkt on the other hand is okay I think: L2 can just ignore it, it's only 2 bytes and I don't think it's a major memory issue for the near future net_pkt memory management changes.

About ED scan, are your sure it's asynchronous? 
Function to implement seems to be: 
`otError otPlatRadioEnergyScan(otInstance *aInstance, u8_t aScanChannel,
u16_t aScanDuration)`

dleach02, 2018-03-16:
On FCS, I'm not suggesting we do the check. I think the OT stack conditionally does this. Maybe they run on some platforms where there is no HW FCS verification. My suggestion here is to change our native stack to accept the whole packet including FCS but just strip off the FCS before passing up. This will eliminate one of the conditional compile code blocks in all of the drivers that support OT and native 15.4.

ED Scan, I just looked at the implementation on KW41Z. The call to otPlatRadioEnergyScan puts the radio into a mode to start scanning and at the same time sets a PHY timer then returns. Note that the return value is just an otError and not an energy level. Later, when the PHY timer expires the ISR is called and it sets a flag for "energy detect done".  

There is an OT process thread that calls a PlatformProcessDrivers() and in PlatformProcessDrivers it calls a radio function to check for TxDone, RxDone, and Energy scan done. On scan done it calls:

otPlatRadioEnergyScanDone(aInstance, sMaxED);

tbursztyka, 2018-03-16:
Sure we won't check FCS ourselves, I was just pointing out that some devices do hw check or not, but that affects the content of the FCS then. I just hope only TI's chip does this and no other ones.

tbursztyka, 2018-03-19:
@KaSroka , @dleach02 : 

Something like that on driver side should be fine then:
`int (*ed_scan)(struct device *dev, u16_t duration, void (*done_cb)(struct device *));`

It would be just up to the OT adaptation layer to set the channel through driver's API set_channel() prior to call that one.

rlubos, 2018-03-22:
Hi!
Are you sure OpenThread conditionally enables to check FCS in the stack? I couldn't recall such a feature, and I've looked through the code of the MAC layer and haven't found anything like this. But as already mention OT expects FCS bytes to be passed. 

@tbursztyka nRF52840 hardware overwrites FCS bytes as well. In our case LQI is put in there.

According ED api, aren't we missing ED result in the callback function? Or is it hidden somewhere inside `device` structure?

KaSroka, 2018-03-22:
@tbursztyka  

@rlubos will take over OpenThread in Zpehyr from me

tbursztyka, 2018-03-22:
@KaSroka ok 

@rlubos So does OT really check the FCS or it just expects to have "something" it does not care about, in place what should be the FCS?

rlubos, 2018-03-22:
To my knowledge - it does not. I've double checked the sources and did not find any place in the stack that would verify it. It expects to get the whole PSDU, but it does not care about the contents of FCS. The only source of `OT_ERROR_FCS` error in OT is the radio driver.

tbursztyka, 2018-03-22:
Ok that facilitates then. Thanks for checking and confirming this

tbursztyka, 2018-05-04:
Please review #7346