zephyrproject-rtos/zephyr #5723: cmake: Accept CONFIG_XX overrides from command line

aescolar-ot, 2018-01-18:
It would be very nice to have the option to override configuration of a project from the command line when calling cmake.
Something like:
```cmake .. -DBOARD=native_posix -DCONFIG_COVERAGE=y -DCONFIG_BT=n```

The idea is that this configuration overrides would have priority over project and arch/board/driver.. options.
They would be fully ephemeral == lost as soon as a new reconfiguration is run.
(If the user wants persistent changes to the configuration he should change the project file)

The 2 uses I see right now:
* Automation: A script like sanitycheck could set CONFIG_COVERAGE from the command line, instead of setting it in the prj files.
* Quick overrides for manual tests to enable/disable features.

pfalcon, 2018-01-18:
I'd find this useful. More specifically, I'd suggest to establish a suitable CFLAGS-like option, e.g. CFLAGS_EXTRA, using which you can pass any $(CC) flags, being them -D or any other.

No idea how this would work with Cmake though.

aescolar-ot, 2018-01-18:
@pfalcon : Note that I meant configuration options. Not just the C pre-processor flags in which they end.

SebastianBoe, 2018-01-18:
> I'd find this useful. More specifically, I'd suggest to establish a suitable CFLAGS-like option, e.g. CFLAGS_EXTRA, using which you can pass any $(CC) flags, being them -D or any other.

cmake -DEXTRA_CFLAGS="-DFOO=3 -fpermissive"

is supported.

pfalcon, 2018-01-18:
@SebastianBoe : any way to pass that into Makefile generated by CMake?

SebastianBoe, 2018-01-19:
> @SebastianBoe : any way to pass that into Makefile generated by CMake?
Assuming you mean passing it when Make is invoked then no.
The closest thing would be

cmake .. -DEXTRA_FLAGS="-DFOO=3" && make

nashif, 2018-01-19:
> @SebastianBoe : any way to pass that into Makefile generated by CMake?

how is this related to this issue? 

I think this is about overriding Kconfig variables on the command line and also resolve dependencies, not just define them.

ulfalizer, 2018-01-19:
If you can get the overrides into `kconfig.py`, then it should be as simple as calling `kconf.syms[sym_name].set_value(value)` after loading the fragments. `set_value()` is what `load_config()` uses internally as well.

That will also generate the `warning: BLAH was assigned the value ... but got the value ...` warning if the symbol gets a different value due to dependencies, and will also warn if the value is invalid for the type or if the symbol has no prompt (user values never have an affect on promptless symbols).

Automatically enabling dependencies is a pretty common request, but unfortunately it's trickier than it might seem at first. Consider a dependency like `depends on (FOO || !BAR) && BAZ != "hmz"` for example. Probably best to require people to manually enable dependencies.

I added myself as a watcher. Might've noticed...

SebastianBoe, 2018-05-08:
Re-assigning to me as this is a build-system feature request.

aescolar, 2018-06-14:
@SebastianBoe : Does this one look to tricky to implement?

PR #8487 (2018-06-20): cmake: Support specifying Kconfig options on the CLI
_NB: This replaces the original #8458 PR (was not able to re-open it)._

This resolves #5723 by adding support for assigning to Kconfig options
from the CMake CLI.

CMake CLI Kconfig options can be specified like so:

'cmake -DCONFIG_ASSERT=y'

The lifetime of such an option is the same as the lifetime of a CMake
cache variable, and in principle any method of populating the
CMakeCache could be used, such as cmake-gui.

This has been implemented by scanning the CMakeCache variables before
Kconfig is executed and writing any that are prefixed  [... cut]
commit 608778a4 (2018-06-20): cmake: Support specifying Kconfig options on the CLI

This resolves #5723 by adding support for assigning to Kconfig options
from the CMake CLI.

CMake CLI Kconfig options can be specified like so:

'cmake -DCONFIG_ASSERT=y'

The lifetime of such an option is the same as the lifetime of a CMake
cache variable, and in principle any method of populating the
CMakeCache could be used, such as cmake-gui.

This has been implemented by scanning the CMakeCache variables before
Kconfig is executed and writing any that are prefixed with 'CONFIG_'
to a new Kconfig fragment in the build directory that is  [... cut]
closed 2018-06-20 by nashif