zephyrproject-rtos/zephyr #5726: CI should use the same generator as the sanitycheck default

SebastianBoe, 2018-01-18:
It is important that it is easy to reproduce CI issues locally. Using
the same sanitycheck options locally and in CI helps in this
regard. Specifically, Ninja and Make can produce different results and
therefore the default generator should be the same for sanitycheck and
shippable.

# Solution proposal
Have sanitycheck default to Ninja as soon as it is stable enough.

SebastianBoe, 2018-01-18:
Blocked by #5725

SebastianBoe, 2018-01-18:
Solution proposal: https://github.com/SebastianBoe/zephyr/commit/8ba5d9329aee1a5ade18e8713da748a1daa82c15

galak, 2018-01-26:
This should also deal with -R.  We should have sanitycheck default to enabling asserts and add an option to explicitly disable.

galak, 2018-01-26:
> This should also deal with -R. We should have sanitycheck default to enabling asserts and add an option to explicitly disable.

The -R part should be addressed by #5841.  Once merged we can update CI to drop -R

PR #6114 (2018-02-12): sanitycheck: make defaults consistent with CI
enable ninja by default and remove -R option from CI file.

Fixes #5726
closed 2018-02-12 by nashif