zephyrproject-rtos/zephyr #5735: mpu_stack_guard_test fails on many arm platforms, including qemu

nashif, 2018-01-19:
sample samples/mpu/mpu_stack_guard_test fails on frdm_k64f:

```
Connecting to /dev/ttyACM0
***** BOOTING ZEPHYR OS v1.10.99 - BUILD: Jan 19 2018 03:15:21 *****
STACK_ALIGN 0x8
MPU STACK GUARD Test
Canary Initial Value = 0xf0cacc1a threads 0x200010e0
Canary = 0x20000240	Test not passed.
***** BUS FAULT *****
  Executing thread ID (thread): 0x200010e0
  Faulting instruction address:  0x209c
  Imprecise data bus error
Fatal fault in thread 0x200010e0! Aborting.
***** HARD FAULT *****
  Fault escalation (see below)
***** BUS FAULT *****
  Executing thread ID (thread): 0x200010e0
  Faulting instruction address:  0x1f4
  Imprecise data bus error
Fatal fault in ISR! Spinning...
```

agross-linaro, 2018-01-26:
Just double checked this.  The expected case when using the default prj file is to fail.  The output given above indicates the prj.conf was used.  If the prf_stack_guard.prj is used, the error's will be correctly determined to be stacking errors.

agross-linaro, 2018-01-26:
It should be noted that qemu (current version in zephyr) does not support MPU.  And not all boards have MPUs.  So perhaps we need to adjust the whitelist.

nashif, 2018-11-05:
AFAIK should be resolved with the new SDK support MPU on ARM.

closed 2018-11-05 by nashif