zephyrproject-rtos/zephyr #5738: [Coverity CID: 182194] Control flow issues in /subsys/storage/flash_map/flash_map.c

Shilpashree3lc, 2018-01-19:
Static code scan issues seen in File: /subsys/storage/flash_map/flash_map.c
 Category: Control flow issues
 Function: flash_map_init
 Component: Other
 CID: 182194
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

Shilpashree3lc, 2018-03-06:
*** CID 182194:  Control flow issues  (NO_EFFECT)
/subsys/storage/flash_map/flash_map.c: 271 in flash_map_init()
265     }
266     
267     static int flash_map_init(struct device *dev)
268     {
269     	int i;
270     
>>>     CID 182194:  Control flow issues  (NO_EFFECT)
>>>     This less-than-zero comparison of an unsigned value is never true. "i < 0UL".

271     	for (i = 0; i < ARRAY_SIZE(flash_dev); i++) {
272     		flash_dev[i] = device_get_binding(flash_drivers_map[i].name);
273     	}
274     
275     	return 0;
276     }
277

carlescufi, 2018-03-06:
@nvlsianpu take a look at the test configuration, it might be that the config is wrong and you are including this file in the build with no flash devices, which should not happen

closed 2018-03-27 by nvlsianpu