zephyrproject-rtos/zephyr #5739: [Coverity CID: 182193] Control flow issues in /subsys/storage/flash_map/flash_map.c

Shilpashree3lc, 2018-01-19:
Static code scan issues seen in File: /subsys/storage/flash_map/flash_map.c
 Category: Control flow issues
 Function: get_flah_dev_from_id
 Component: Other
 CID: 182193
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

Shilpashree3lc, 2018-03-06:
*** CID 182193:  Control flow issues  (NO_EFFECT)
/subsys/storage/flash_map/flash_map.c: 74 in get_flah_dev_from_id()
68     {
69     	/* nothing to do for now */
70     }
71     
72     static struct device *get_flah_dev_from_id(u8_t id)
73     {
>>>     CID 182193:  Control flow issues  (NO_EFFECT)
>>>     "i" is converted to an unsigned type because it's compared to an unsigned constant.

74     	for (int i = 0; i < ARRAY_SIZE(flash_drivers_map); i++) {
75     		if (flash_drivers_map[i].id == id) {
76     			return flash_dev[i];
77     		}
78     	}
79

carlescufi, 2018-03-06:
@nvlsianpu can you use `unsigned int` here?

closed 2018-03-27 by nvlsianpu