zephyrproject-rtos/zephyr #5742: [Coverity CID: 181921] Incorrect expression in /subsys/bluetooth/controller/ll_sw/ctrl.c

Shilpashree3lc, 2018-01-19:
Static code scan issues seen in File: /subsys/bluetooth/controller/ll_sw/ctrl.c
 Category: Incorrect expression
 Function: role_active_disable
 Component: Bluetooth
 CID: 181921
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

Shilpashree3lc, 2018-03-06:
*** CID 181921:  Incorrect expression  (ASSERT_SIDE_EFFECT)
/subsys/bluetooth/controller/ll_sw/ctrl.c: 9506 in role_active_disable()
9500     			  ticker_if_done, (void *)&ret_cb_evt);
9501     
9502     	if (ret == TICKER_STATUS_BUSY) {
9503     		mayfly_enable(RADIO_TICKER_USER_ID_APP,
9504     			      RADIO_TICKER_USER_ID_JOB, 1);
9505     
>>>     CID 181921:  Incorrect expression  (ASSERT_SIDE_EFFECT)
>>>     Argument "ret_cb_evt" of LL_ASSERT() has a side effect because the variable is volatile.  The containing function might work differently in a non-debug build.

9506     		LL_ASSERT(ret_cb_evt != TICKER_STATUS_BUSY);
9507     	}
9508     
9509     	if (ret_cb_evt == TICKER_STATUS_SUCCESS) {
9510     		static memq_link_t link_xtal;
9511     		static struct mayfly s_mfy_xtal_stop = {0, 0, &link_xtal, NULL,

carlescufi, 2018-03-06:
@cvinayak can we get this fixed for 1.11 (Friday)?

carlescufi, 2018-05-04:
Intentional, commented in Coverity

closed 2018-05-04 by carlescufi