zephyrproject-rtos/zephyr #5743: Windows and Linux write .config files with options re-ordered

SebastianBoe, 2018-01-19:
Native Windows and Linux are not producing identical .config files, but AFAICT they are at least
producing semantically equivalent files.

The order in which options are written into the file is different. This complicates the process of verifying that two builds are equivalent so we need to find out why and fix this.

Steps to reproduce:

Run the below command on each platform and then diff the build/zephyr/.config file.
~/zephyr/samples/hello_world$ cmake -DBOARD=nrf52_pca10040 -H. -Boe -GNinja

Tested on master commit 812e2eaad28f288556398237c01f021a89cc0f2c

ulfalizer, 2018-01-19:
Hello,

Symbols appear in the same order in the generated `.config` files as they do in the `Kconfig` files.

If the `.config` files are generated by Kconfiglib, then I suspect that the difference is due to the custom `glob()` logic for `source`. Python's `glob.glob()` returns the list of files in an arbitrary order (unlike the POSIX `glob()`, which sorts them).

This patch might fix it:

```diff
diff --git a/scripts/kconfig/kconfiglib.py b/scripts/kconfig/kconfiglib.py
index fdaa7a2b1..ec0c6d04b 100644
--- a/scripts/kconfig/kconfiglib.py
+++ b/scripts/kconfig/kconfiglib.py
@@ -1074,7 +1074,7 @@ class Kconfig(object):
 
         if globbing:
             # Try globbing
-            return glob.glob(filename)
+            return sorted(glob.glob(filename))
 
         raise IOError(
             "Could not find '{}'. Perhaps the $srctree "
```

SebastianBoe, 2018-01-19:
Thanks @ulfalizer , I'll give that a try.

PR #6079 (2018-02-09): kconfig: Sort the glob results
Sort the glob results when Kconfig sources do 'source
"/path/*/Kconfig"' to ensure that the resulting dotconfig file is the
same for all platforms.

This fixes #5743

Credit goes to @ulfalizer.
commit 6bb98cc4 (2018-02-09): kconfig: Sort the glob results

Sort the glob results when Kconfig sources do 'source
"/path/*/Kconfig"' to ensure that the resulting dotconfig file is the
same for all platforms.

This fixes #5743

Credit goes to @ulfalizer.
closed 2018-02-09 by nashif