zephyrproject-rtos/zephyr #5749: Exception and Interrupt vector forwarding

carlescufi, 2018-01-19:
There's been some confusion regarding the requirements and scope of this issue so I will try to summarize here what (I think) needs to be done in order to be able to use mcuboot with Zephyr on platforms that do not support vector table relocation in hardware.

Glossary:
* bootloader image: The bootloader binary image, always at the beginning of flash
* application image: The application binary image that needs to be booted by the bootloader, placed in flash after the Bootloader image
* vector table: The hardware vector table that the CPU uses to find an exception or interrupt handler 

Let's describe the issue:

* On most variants of the ARM Cortex-M CPU there is a way to relocate the vector table in hardware. This is typically necessary when using a bootloader and, in Cortex-M3 and M4 this is achieved through the use of the `VTOR` register, which is able to "offset" the vector table in hardware , pointing to the one in  the application image. This is done just before entering the C code, [here](https://github.com/zephyrproject-rtos/zephyr/blob/master/arch/arm/core/cortex_m/prep_c.c#L46).

* On most variants of the ARM Cortex-M0 CPU however, the CPU lacks this `VTOR` register, and this is an issue since we need to tell the CPU to use the vector table from the application image once it's booted, not the one from the bootloader image.

* Both the bootloader image (when using mcuboot with Zephyr) and the application image (when using Zephyr) use the same boot process and exception and interrupt handling tables and mechanism, the ones from Zephyr itself

* In Zephyr, the vector table is placed in the following linker symbols:
```
_vector_start # Start of the vector table, always 0x0 for mcuboot
_vector_table # The hardware exception vector table, including reset vector and initial stack pointer
_irq_vector_table # The hardware interrupt vector table
_sw_isr_table # The software interrupt vector table, used by _isr_wrapper to forward the interrupt
_vector_end # End of the vector table
```
* The main issue is that the area between `_vector_start` and `_vector_end` in mcuboot is placed in **flash memory** and not RAM. This means that we cannot overwrite it easily, using something like [this](https://github.com/zephyrproject-rtos/zephyr/blob/master/arch/arm/core/cortex_m/prep_c.c#L30), we would have to actually modify the bootloader image in flash, and that is not a reasonable option. Otherwise it would be simple and we would just `memcpy` into it like the function shows.

* Given all the facts above, we need a software mechanism to relay or forward the interrupts handled by the bootloader image's hardware vector table in flash into the application image's vector table in flash, which cannot be reached by the CPU without software help.

One option is to have an additional level of indirection that is always present, in the form of a relay vector table but only in the bootloader image, since the application image does not need to forward. This relay vector table contains all its exceptions and interrupt pointers pointing to a single relay ISR that is able to forward correctly to the application's actual vector table in flash. In order for this to work, the bootloader image's relay ISR needs to know the location of the application image's vector table in flash in order to jump into it, and must do so only after the application image has booted. Before then the bootloader image's relay vector table needs to relay the exceptions and interrupts to the bootloader image's vector table.

The pull request proposed by @miyatsu [here](https://github.com/zephyrproject-rtos/zephyr/pull/4536) attempts to implement this approach, but I think a couple of items are missing:

1) I don't see how the bootloader image's relay vector table knows about the application image's vector table address in flash
2) I don't see how the bootloader image's relay vector table is correctly switching from forwarding the exceptions and interrupts to the bootloader image's vector table to the application image's vector table.

carlescufi, 2018-01-19:
@ioannisg @mbolivar @miyatsu @d3zd3z @andrewboie FYI

miyatsu, 2018-01-22:
Hi, thank you for the summarize.

1).MCUboot always try to load reset value of application image from flash address: "FLASH_AREA_IMAGE_0_OFFSET" + "CONFIG_TEXT_SECTION_OFFSET". The 1st value "FLASH_AREA_IMAGE_0_OFFSET" is generated from dts; and the 2nd value "CONFIG_TEXT_SECTION_OFFSET" is the size of reserved space for image signature, suggest 0x200 for most target/plantform.

So I want to changed the zephyr's linker script to adapt it, put the vector table at the very beginning of output image file.
At this point, for Cortex-M0 target, the irq relay handler will know how to find the application's vector table.

2).There is no operating system supported on this IRQ relay handler, it works independently. The IRQ relay mechanism works like this:

An IRQ coming, it will first processed by the IRQ relay handler, the IRQ relay handler will calculate is this IRQ for bootloader or for application. (Actually, IRQ relay handler don't have to calculate it. We assume the vector table will be copy to address 0x20000000, if an application image is booted, the application will rewrite that little piece of RAM use its own vector table.)

Then the IRQ relay handler just jump to the correct IRQ handle address, pass the control to what ever it is, bootloader or application, IRQ relay handler won't care.

nashif, 2018-01-30:
done

closed 2018-01-30 by nashif