zephyrproject-rtos/zephyr #5762: Running a Sample Application natively (POSIX OS)

xianrenqiu, 2018-01-20:
I refer to [《Native POSIX execution (native_posix)》](http://docs.zephyrproject.org/boards/posix/native_posix/doc/board.html) chapter，When I execute zephyr/zephyr.exe command.  An Error Has Occurred：
> **-bash: zephyr/zephyr.exe: cannot execute binary file: Exec format error**

the process is as follows：
```
$ cd $ZEPHYR_BASE/samples/hello_world
$ mkdir build && cd build
$ cmake -DBOARD=native_posix ..
$ make
$ sudo zephyr/zephyr.exe
-bash: zephyr/zephyr.exe: cannot execute binary file: Exec format error
```
I use the ubuntu 16.04(64 bit) operating system. and i have already install the gcc-multilib package.

What might be the problem? Thank you.

xianrenqiu, 2018-01-21:
I found it was a [WSL problem](https://github.com/Microsoft/WSL/issues/390)，Currently WSL only supports amd64 (native 64bit) ，does not supports 32-bit i386 ELF binaries 

> Whether we can add relevant explanations in [relevant chapters](http://docs.zephyrproject.org/getting_started/installation_win.html#using-windows-10-wsl-windows-subsystem-for-linux).

aescolar-ot, 2018-01-21:
Thanks, I never expected it to work in WSL, you may have been the 1st one to try it out.

@carlescufi Will you consider @xianrenqiu request on the warning in the WSL installation chapter?

nashif, 2018-02-06:
no plans to support this in WSL since it only supports 64bit.

closed 2018-02-06 by nashif