zephyrproject-rtos/zephyr #5766: boards: nucleo_f413zh: Likely outdated OpenOCD info in docs

pfalcon, 2018-01-20:
boards/arm/nucleo_f413zh/doc/nucleof413zh.rst says:

~~~
Nucleo F413ZH board includes an ST-LINK/V2-1 embedded debug tool interface.
However this interface is currently not supported by OpenOCD. You will need
to use ST tools or an external JTAG probe.
~~~

Most other STM32 boards say:

~~~
Nucleo L432KC board includes an ST-LINK/V2-1 embedded debug tool
interface.  This interface is supported by the openocd version
included in the Zephyr SDK since v0.9.2.
~~~

Perhaps, nucleo_f413zh docs need to be updated?

pfalcon, 2018-01-20:
Same for nucleof429zi.rst.

pfalcon, 2018-01-20:
Also nucleo412zg.rst

galak, 2018-02-06:
@erwango any comments on this?

erwango, 2018-02-08:
Well, I think this is indeed incorrect information.
I've been able to check flash is successful on nucleo_f429zi and nucleo_f412zg and I don't see why it shouldn't work on last one.
I'll fix this.

PR #6057 (2018-02-08): Fix nucleo flash support
Some nucleo boards were missing config files for openocd support.
Fix this and update doc

Fixes #5766
closed 2018-02-08 by galak