zephyrproject-rtos/zephyr #5767: docs: Zephyr OpenOCD documentation is weal/non-existent

pfalcon, 2018-01-20:
Quite a number of boards supported by Zephyr rely on OpenOCD. However, there's barely any generic information how OpenOCD can be used/controlled/is integrated into Zephyr toolset/build system.

The only information available are copy-paste flashing and debugging instructions, e.g. http://docs.zephyrproject.org/boards/arm/disco_l475_iot1/doc/disco_l475_iot1.html#flashing . They don't go any further than saying "ninja flash". As an outstanding exception, http://docs.zephyrproject.org/boards/arm/sam_e70_xplained/doc/sam_e70_xplained.html#flashing actually mentions that there's `OPENOCD_INTERFACE` envvar which can be controlled one aspect of OpenOCD functioning.

Beyond that, there's absolutely no generic, board-independent documentation on OpenOCD in Zephyr. That's unlike e.g. QEMU and gdb, which have nice, generic quick start instructions to let users easily going (but also include some additional usage and troubleshooting hints).

pfalcon, 2018-01-20:
@mbolivar : I wonder if you would agree with evaluation above.

@nashif, @dbkinder : Wonder if you think it makes to address the above.

mbolivar, 2018-01-29:
@pfalcon I'm never against improving documentation. What is the concern here, though? Is it setting up development environments on OS X or Windows?

Since on Linux, the SDK provides something which "just works" in terms of the flash/debug/debugserver targets, I'm not sure what additional Zephyr-specific documentation about openocd you're looking for.

pfalcon, 2018-01-29:
@mbolivar , Thanks for the reply.

> Is it setting up development environments on OS X or Windows?

Well, these ones are never my concern ;-).

> Since on Linux, the SDK provides something which "just works" in terms of the flash/debug/debugserver 

What if it doesn't? For example, when did you last time ran openocd without a JTAG adapter attached? It just terminates in this case, without any indication that there's an error. And just imagine that between these polar situations "it just works" and "it can't work 100%, and doesn't even tell why it doesn't" there're gazillion of situation where user would be lost why it doesn't work and what to do about that.

> I'm not sure what additional Zephyr-specific documentation about openocd you're looking for.

I don't look for "Zephyr-specific documentation" per se. As mentioned above, I'd rather look for a "cheat sheet", where examples (if needed, and usually they are) are given as Zephyr specific. I'd use GDB  http://docs.zephyrproject.org/application/application.html#application-debugging and QEMU networking http://docs.zephyrproject.org/subsystems/networking/qemu_setup.html chapters as examples of what I mean. They provide a good overview of how to get to work with those tools, definitely more than just "run 'make flash'".

That said, this ticket is a "reality check" whether people see the problem ("openocd in particular and JTAG and friends in general are one of the most confusing topics in embedded development") which needs to be resolved, or if it's my (mis)perception. I definitely don't want to spend time on something which has another/better solution (like RTFM some other manual(s)), or make Zephyr docs be a collection of my personal (mis)conceptions ;-).

pfalcon, 2018-01-30:
> Are you looking for docs on Zephyr's runners, with pages for each of the supported back-ends, including openocd? I'd be amenable to writing some, though I'm not ready to freeze the command line interfaces yet.

Not really, or more specifically, that's not what this topic is about. The idea is that majority of boards can be programmed/flashes using JTAG or SWD interface, and the most advanced tool we have to control that is OpenOCD (I'd really much like that to be pyOCD, but surely, we aren't yet there/probably won't be there (not JTAG support in pyOCD, right?)). OpenOCD is however a rather esoteric and confusing tool, and I personally would look for a "cheat sheet" how to to do basic operations and troubleshooting with it.

All in all, I see that my idea doesn't ring a bell. Let me assign this to myself and set priority as low, and keep collecting my OpenOCD experiences (and/or maybe collect more feedback over time).

Thanks!

pfalcon, 2018-05-29:
Let this be for now a personal opinion, and let's close it then (for now).

closed 2018-05-29 by pfalcon