zephyrproject-rtos/zephyr #5771: Linking issues with host cross compile with cmake

nullr0ute, 2018-01-22:
Related to issue #5589 with the migration to cmake there seems to be some issues around linking with the new cmake. I checked against 1.9 in case it was a host cross compile stack regression.

With PR #5594 merged it gets a lot further than it did with the following commands:

<pre>
export CROSS_COMPILE="/usr/bin/arm-linux-gnu-"
export ZEPHYR_GCC_VARIANT=cross-compile
unset ZEPHYR_SDK_INSTALL_DIR
source zephyr-env.sh

cd samples/hello_world/
mkdir build && cd build
cmake -DCMAKE_VERBOSE_MAKEFILE=1 -DBOARD=96b_carbon ..
make V=1
<pre>

But fails at the linking stage because it seems to be trying to use the Cortex-A options:

<pre>
[ 95%] Linking C executable zephyr_prebuilt.elf
cd /home/perobins/code/zephyr/samples/hello_world/build/zephyr && /usr/bin/cmake -E cmake_link_script CMakeFiles/zephyr_prebuilt.dir/link.txt --verbose=1
/usr/bin/arm-linux-gnu-gcc    CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr_prebuilt.elf -T linker.cmd -Wl,-Map=/home/perobins/code/zephyr/samples/hello_world/build/zephyr/zephyr.map -u_OffsetAbsSyms -u_ConfigAbsSyms -e__start -Wl,--start-group -Wl,--whole-archive ../libapp.a libzephyr.a boards/arm/stm32f3_disco/libboards__arm__stm32f3_disco.a -Wl,--no-whole-archive kernel/libkernel.a CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -Wl,--end-group -L"/usr/lib/gcc/arm-linux-gnueabi/7" -L/home/perobins/code/zephyr/samples/hello_world/build/zephyr -lgcc -Wl,--print-memory-usage -nostartfiles -nodefaultlibs -nostdlib -static -no-pie -Wl,-X -Wl,-N -Wl,--gc-sections -Wl,--build-id=none 
Memory region         Used Size  Region Size  %age Used
           FLASH:       11076 B       256 KB      4.23%
            SRAM:        4772 B        40 KB     11.65%
        IDT_LIST:         132 B         2 KB      6.45%
/usr/bin/arm-linux-gnu-ld: error: /usr/lib/gcc/arm-linux-gnueabi/7/libgcc.a(_aeabi_ldivmod.o): Conflicting architecture profiles A/M
/usr/bin/arm-linux-gnu-ld: failed to merge target specific data of file /usr/lib/gcc/arm-linux-gnueabi/7/libgcc.a(_aeabi_ldivmod.o)
/usr/bin/arm-linux-gnu-ld: error: /usr/lib/gcc/arm-linux-gnueabi/7/libgcc.a(_aeabi_uldivmod.o): Conflicting architecture profiles A/M
/usr/bin/arm-linux-gnu-ld: failed to merge target specific data of file /usr/lib/gcc/arm-linux-gnueabi/7/libgcc.a(_aeabi_uldivmod.o)
/usr/bin/arm-linux-gnu-ld: error: /usr/lib/gcc/arm-linux-gnueabi/7/libgcc.a(_dvmd_lnx.o): Conflicting architecture profiles A/M
/usr/bin/arm-linux-gnu-ld: failed to merge target specific data of file /usr/lib/gcc/arm-linux-gnueabi/7/libgcc.a(_dvmd_lnx.o)
/usr/bin/arm-linux-gnu-ld: error: /usr/lib/gcc/arm-linux-gnueabi/7/libgcc.a(_udivmoddi4.o): Conflicting architecture profiles A/M
/usr/bin/arm-linux-gnu-ld: warning: /usr/lib/gcc/arm-linux-gnueabi/7/libgcc.a(_udivmoddi4.o) uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
/usr/bin/arm-linux-gnu-ld: failed to merge target specific data of file /usr/lib/gcc/arm-linux-gnueabi/7/libgcc.a(_udivmoddi4.o)
collect2: error: ld returned 1 exit status
make[2]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:105: zephyr/zephyr_prebuilt.elf] Error 1
make[2]: Leaving directory '/home/perobins/code/zephyr/samples/hello_world/build'
make[1]: *** [CMakeFiles/Makefile2:545: zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
make[1]: Leaving directory '/home/perobins/code/zephyr/samples/hello_world/build'
</pre>

I don't think this is an issue with the Fedora cross toolchain as it still works with the 1.9 branch. I'm no expert with cmake/linker flags but I think these might have changed in the move to cmake, and I suspect there's a difference in the defaults between the zephyr toolchain vs the host toolchain.

SebastianBoe, 2018-01-22:
Thank you for the report.

The linker command itself looks good to me.

It looks like this with the Zephyr SDK:

> /home/sebo/zephyr-sdk-0.9.2-rc5/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gcc
> zephyr/CMakeFiles/kernel_elf.dir/misc/empty_file.c.obj
> zephyr/CMakeFiles/kernel_elf.dir/isr_tables.c.obj -o
> zephyr/zephyr.elf -T zephyr/linker_pass2.cmd
> -Wl,-Map=/home/sebo/zephyr/samples/hello_world/b/zephyr/zephyr.map
> -u_OffsetAbsSyms -u_ConfigAbsSyms -e__start -Wl,--start-group
> -Wl,--whole-archive libapp.a zephyr/libzephyr.a
> zephyr/boards/arm/96b_carbon/libboards__arm__96b_carbon.a
> -Wl,--no-whole-archive zephyr/kernel/libkernel.a
> zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
> -Wl,--end-group
> -L"/home/sebo/zephyr-sdk-0.9.2-rc5/sysroots/armv5-zephyr-eabi/usr/lib/arm-zephyr-eabi/6.2.0/armv7e-m"
> -L/home/sebo/zephyr/samples/hello_world/b/zephyr -lgcc
> -nostartfiles -nodefaultlibs -nostdlib -static -no-pie -Wl,-X
> -Wl,-N -Wl,--gc-sections -Wl,--build-id=none
> 

I think it is reasonable to assume the Fedora cross toolchain is OK as well.

If I assume that /usr/lib/gcc/arm-linux-gnueabi/7/libgcc.a is a cortex-m binary (you can verify if unsure with readelf -a /usr/lib/gcc/arm-linux-gnueabi/7/libgcc.a | grep Tag_CPU), then it must be one of the object files that are pulling in libgcc.a as a dependency that has been incorrectly compiled as Cortex-A.

If you are inclined to do so, please dump a verbose build so I can see how it has been compiled.

e.g.

"make V=1"

Specifically, this code:
https://github.com/zephyrproject-rtos/zephyr/blob/master/cmake/toolchain-gcc.cmake#L58
should have ensured that -mthumb and -mcpu=cortex-m4 were used when compiling object files (but not when linking).

nullr0ute, 2018-01-22:
Complete verbose build output:
[
[zephyr-build-output.txt](https://github.com/zephyrproject-rtos/zephyr/files/1651804/zephyr-build-output.txt)
](url)

nashif, 2018-02-20:
what is the status of this?

SebastianBoe, 2018-02-21:
Looks like a bug, but the issue has not been reproduced and the root cause is unknown.

nullr0ute, 2018-02-21:
Did you see anything in my attached logs? Can I check anything else? I've probably got some cycles this week to do more tests for you

SebastianBoe, 2018-02-23:
Hi,

This error:

"/usr/bin/arm-linux-gnu-ld: error: /usr/lib/gcc/arm-linux-gnueabi/7/libgcc.a(_aeabi_ldivmod.o): Conflicting architecture profiles A/M"

Tells me that the linker is been given conflicting information on whether to link a Cortex-A or Cortex-M binary.

Now, we do not specify to the linker itself what architecture to link for and (unless this is proving me wrong) we do not have to. It infers it from the binaries given to it. Finding out what is going wrong is then simply a matter of checking each binary given to the linker and checking that they are all the same architecture.

To check this you can run this command:

readelf -a $BINARY | grep Tag_CPU

on each binary given to the linker. So:

/usr/lib/gcc/arm-linux-gnueabi/7/libgcc.a
CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj
../libapp.a
libzephyr.a
boards/arm/96b_carbon/libboards__arm__96b_carbon.a
 kernel/libkernel.a
CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj

SebastianBoe, 2018-02-23:
My best guess is that "/usr/lib/gcc/arm-linux-gnueabi/7/libgcc.a" is a Cortex-A v7, and not a Cortex-M v7 binary, because AFAICT all the Zephyr sources were built with -mcpu=Cortex-M4.

nullr0ute, 2018-02-23:
The zephyr 1.9 release/branch builds OK using the same toolchain so there's some variation in how we're doing it in cmake

SebastianBoe, 2018-02-26:
Hi,

before I set up a Docker image to reproduce this ...

I see that the toolchain used is named like this:

"export CROSS_COMPILE="/usr/bin/arm-linux-gnu-"

and your guide specifies this package:

"gcc-arm-linux-gnu"

Which documents itself like this:

"Cross-build GNU C compiler. Only building kernels is currently supported."

would you mind seeing if this fedora package, or this RPM (not familiar with Fedora) has the same issues?

https://apps.fedoraproject.org/packages/arm-none-eabi-gcc-cs
http://rpmfind.net/linux/rpm2html/search.php?query=arm-none-eabi-gcc

If one of them works we will have narrowed it down to there being a compatibility issue with this specific package, and not CROSS_COMPILE in general.

https://rpmfind.net/linux/rpm2html/search.php?query=gcc-arm-linux-gnu

PS: I'm not familiair with the package, but is "kernels" in "Only building kernels is currently supported." meant to communicate that it is only supported to build the Linux kernel?

nullr0ute, 2018-02-26:
The "Only building kernels is currently supported" basically means kernels, firmwares and other such things. It doesn't deal with dynamic libraries etc.

I use this toolchain to build zephyr 1.9 without issues so I'm pretty sure it's not the toolchain, also use it to build u-boot, arm-trusted-firmware, micropython for micro controllers and other similar things.

It actually compiles the master branch fine, it has issues at the linking stage, I don't believe this is the toolchain as it's been used to compile eg the 96b_carbon in the past just fine

SebastianBoe, 2018-02-26:
Right, so I set up a Fedora docker image and can confirm by running the readelf test I described earlier that link is failing because

/usr/lib/gcc/arm-linux-gnueabi/7/libgcc.a

is a Cortex-A library, and not a Cortex-M library. Which beg's the question, why is the Zephyr buildsystem using a Cortex-A library. Will find out ...

SebastianBoe, 2018-02-26:
I see that libgcc.a is the only libc implementation provided by "gcc-arm-linux-gnu" which means that Zephyr is not detecting the wrong libc implementation. The package does not provide a Cortex-M libc implementation at all. I can not explain that this toolchain has been used to build Zephyr in the past, I do not see any evidence that it is able to build for Cortex-M.

Following this Fedora wiki for targeting ARM embedded gives the alternative package:

arm-none-eabi-gcc-cs

https://fedoraproject.org/wiki/Embedded#Bare_metal_ARM_without_any_operating_system

which I can confirm is able to build for 96b_carbon on Fedora.

Docker file:

```
FROM fedora

RUN useradd -m zephiroth
USER zephiroth
WORKDIR /home/zephiroth

USER root
RUN yes | dnf install git-core gcc glibc-static libstdc++-static make dfu-util dtc python3-PyYAML

# RUN apt-get update && \
#     apt-get upgrade -y && \
#     apt-get install -y --no-install-recommends \
# 	doxygen \
#     device-tree-compiler \
#     dfu-util \
#     git \
#     gperf \
#     ninja-build \
#     python3-pip \
#     python3-ply \
#     python3-setuptools \
#     file \
#     xz-utils
    
RUN yes | dnf install cmake
    
USER zephiroth

# Install the specific version of CMake that Zephyr supports
ENV CMAKE_VERSION_FULL 3.8.2
ENV CMAKE_VERSION_SHORT 3.8

# RUN yes | sh cmake-$CMAKE_VERSION_FULL-Linux-x86_64.sh | cat
# ENV PATH /home/zephiroth/cmake-$CMAKE_VERSION_FULL-Linux-x86_64/bin:$PATH

RUN git clone https://github.com/zephyrproject-rtos/zephyr.git

RUN pip3  install --user wheel
RUN pip3  install --user -r zephyr/scripts/requirements.txt

ENV SDK_VERSION 0.9.2-rc4
# ADD https://github.com/zephyrproject-rtos/meta-zephyr-sdk/releases/download/$SDK_VERSION/zephyr-sdk-$SDK_VERSION-setup.run $HOME
#ADD zephyr-sdk-$SDK_VERSION-setup.run $HOME

USER root
# RUN chown -R zephiroth:zephiroth zephyr-sdk-$SDK_VERSION-setup.run
USER zephiroth

# RUN sh $HOME/zephyr-sdk-$SDK_VERSION-setup.run -- -d $HOME/zephyr-sdk

ENV ZEPHYR_GCC_VARIANT cross-compile
#ENV CROSS_COMPILE /usr/bin/arm-linux-gnu-
ENV CROSS_COMPILE /usr/bin/arm-none-eabi-
ENV ZEPHYR_BASE /home/zephiroth/zephyr

WORKDIR $ZEPHYR_BASE

USER root
RUN yes | dnf install gperf
USER zephiroth

USER root
RUN yes | dnf install arm-none-eabi-gcc-cs
USER zephiroth

RUN cd samples/hello_world && cmake -DBOARD=96b_carbon -H. -Boe

# && make -C oe

```

I would like to close this as not-a-bug until it has been proven somehow that gcc-arm-linux-gnu can build for Cortex-M (with a toolchain-provided libc implementation).

closed 2018-02-26 by SebastianBoe