zephyrproject-rtos/zephyr #5778: Add/fix flash controller nodes for NXP kinetis SoCs

galak, 2018-01-22:
Its looks like the only flash controller node that might be correct is dts/arm/nxp/nxp_k6x.dtsi.

Several dtsi don't have flash controller nodes at all:
* dts/arm/nxp/nxp_kl25z.dtsi
* dts/arm/nxp/nxp_kw40z.dtsi
* dts/arm/nxp/nxp_kw41z.dtsi

And dts/arm/nxp/nxp_kw2xd.dtsi has copied the k64 node.

Need to update all these .dtsi to have proper flash controller nodes.

galak, 2018-01-25:
It looks like kw2xd is probably reasonable to be copied from k64.  Only difference is in # of cache tags & cache data tags.

galak, 2018-01-25:
Added write-block-size to all dt's based on FSL_FEATURE_FLASH_PFLASH_BLOCK_WRITE_UNIT_SIZE.  This is in PR #5830.

nashif, 2018-02-06:
fixed

closed 2018-02-06 by nashif