zephyrproject-rtos/zephyr #5781: Initial TLS connection failure causes TLS client handler to stop and fail endlessly w/ EBUSY

mike-scott, 2018-01-22:
The TLS startup code looks like this:
net_app_connect() -> start_tls_client()  starts the tls_client_handler thead.

IE: the only way to start the TLS client handler thread is a call to net_app_connect.

If the initial TLS connection fails for any reason (connectivity or otherwise), the TLS client handler is stopped, and using the net_app_connect() call again generates errors as the thread isn't cleared up correctly.

Further, I'm not sure if this is the API flow that we want sample apps to use for making sure TLS comes up correctly.  There isn't much in the way of documentation in this area.

mike-scott, 2018-02-13:
Hi @jukkar 

The easiest way to reproduce the EBUSY is to build the lwm2m_client sample configured for DTLS for qemu_x86 (in master branch) and connect it to a Leshan demo server (LwM2M) server without setting up any DTLS information:

[terminal window #1: run Leshan demo server]
wget https://hudson.eclipse.org/leshan/job/leshan/lastSuccessfulBuild/artifact/leshan-server-demo.jar
java -jar ./leshan-server-demo.jar

[terminal window #2: build / run lwm2m_client for qemu_x86 using DTLS]
cd <zephyr base>/samples/net/lwm2m_client
mkdir build && cd build
cmake -DBOARD=qemu_x86 -DCONF_FILE=prj_dtls.conf ..
make run

You should see a series of -16 (EBUSY) errors and then -12 (ENOMEM) once the heap is used.

jukkar, 2018-02-14:
> You should see a series of -16 (EBUSY) errors and then -12 (ENOMEM) once the heap is used.

Ok, thanks for the steps, I will try to reproduce this.

jukkar, 2018-05-14:
@mike-scott There has been lot of activity with lwm2m code recently, can you confirm if this is still a valid issue?

mike-scott, 2018-05-14:
@jukkar I don't think any of the LwM2M changes would affect the ability to tear down and rebuild the TLS connection in the net_app layer.  I'll retest and post my findings.

nashif, 2018-08-26:
@mike-scott ping

mike-scott, 2018-08-27:
@nashif  Not sure if we want to close this.  It won't be addressed using the net-app APIs.  I'll deal with it in the socket re-write.

nashif, 2018-08-27:
so 1.14?

mike-scott, 2018-08-27:
done

mike-scott, 2019-02-06:
Move to sockets eliminates this bug.

closed 2019-02-06 by mike-scott