zephyrproject-rtos/zephyr #5784: make rom_report fails for qemu_x86 (not finding zephyr.bin)

GAnthony, 2018-01-23:
Previously, rom_report and ram_report targets worked for the mqtt_publisher example.
Now, it seems these targets are broken (or, is there a new usage?)

```
build[master]> cd samples/net/mqtt_publisher/
build[master]> mdkir build; cd build
build[master]> cmake .. -DBOARD=qemu_x86
build[master]> make rom_report
[...]
[100%] Building C object zephyr/CMakeFiles/kernel_elf.dir/misc/empty_file.c.obj
[100%] Linking C executable zephyr.elf
Generating files from zephyr.elf for board: qemu_x86
[100%] Built target kernel_elf
Scanning dependencies of target rom_report
Traceback (most recent call last):
  File "/home/gpitney/zephyr/scripts/footprint/size_report", line 356, in <module>
    0, 0)
  File "/home/gpitney/zephyr/scripts/footprint/size_report", line 94, in get_footprint_from_bin_and_statfile
    total_used_flash = os.path.getsize(bin_file)
  File "/usr/lib/python3.4/genericpath.py", line 50, in getsize
    return os.stat(filename).st_size
FileNotFoundError: [Errno 2] No such file or directory: '/home/gpitney/zephyr/samples/net/mqtt_publisher/build/zephyr/zephyr.bin'
zephyr/cmake/reports/CMakeFiles/rom_report.dir/build.make:57: recipe for target 'zephyr/cmake/reports/CMakeFiles/rom_report' failed
make[3]: *** [zephyr/cmake/reports/CMakeFiles/rom_report] Error 1
CMakeFiles/Makefile2:2582: recipe for target 'zephyr/cmake/reports/CMakeFiles/rom_report.dir/all' failed
make[2]: *** [zephyr/cmake/reports/CMakeFiles/rom_report.dir/all] Error 2
CMakeFiles/Makefile2:2589: recipe for target 'zephyr/cmake/reports/CMakeFiles/rom_report.dir/rule' failed
make[1]: *** [zephyr/cmake/reports/CMakeFiles/rom_report.dir/rule] Error 2
Makefile:664: recipe for target 'rom_report' failed
make: *** [rom_report] Error 2

build[master]> find . -name 'zephyr.bin'
build[master]> 
```

SebastianBoe, 2018-01-23:
This is a regression from https://github.com/zephyrproject-rtos/zephyr/commit/e51ce4d34d3c80c4826eef80d9b9573261d1ee8e

This could be fixed quickly by Kconfig defaulting to generating .bin and .stat files.

Or properly by having .bin and .stat files exist as nodes in the dependency graph instead of as a 'POST_BUILD' command and expressing that the .bin file depends on the .elf file and rom_report depends on the .bin file (same for .stat).

pfalcon, 2018-01-23:
Disabling .bin output for some targets (*all* QEMUs) caused me a bunch of trouble adjusting external integrations and CIs (JerryScript, MicroPython build systems, JerryScript Travis CI inetgration, Linaro CIs for all of Zephyr, JerryScript, MicroPython). I really wanted to submit a ticket with the usual "please don't make random changes breaking user stuff", but contained myself and heroically made that cascade of patches (which of course wasn't that hard overall, but it was that cascade).

GAnthony, 2018-01-23:
SebastianBoe: OK, thanks, with that information, I am at least able to generate the reports by setting CONFIG_BUILD_OUTPUT_BIN=y in my local prj.conf file, for now, until the best solution is taken.

pfalcon, 2018-01-26:
> qemu does not need a bin file to work, we have been generating lots of files

We still generate lots of files. For example, we generate .lst which "not needed", but I find it a big convenience. Likewise, .bin wasn't needed for qemu, but *could* be used, which was a convenience in some circumstances.

All in all, the reasons you listed is why I apparently didn't submit that as an issue. My point, that "didn't help", as other people hit more issues rooted in that change, only later. (So, not reporting issues and just working them around "doesn't help" in the big picture.)

PR #5838 (2018-01-28): scripts: fix ram_reports and generate bin file if missing
Some boards do not generate .bin files by default, this file is however
needed when generating ram/rom reports, so in case it is not present, create
it.

Fixes #5784
commit de311407 (2018-01-28): scripts: fix ram_reports and generate bin file if missing

Some boards do not generate .bin files by default, this file is however
needed when generating ram/rom reports, so in case it is not present,
create it.

Fixes #5784
closed 2018-01-28 by nashif