zephyrproject-rtos/zephyr #5794: wiki/Development-Model is out of date

galak, 2018-01-23:
references JIRA & gerrit

dbkinder, 2018-01-23:
I've done a first pass changing references to Jira and Gerrit and updating the images accordingly.  @galak can you look through the instructions at https://github.com/zephyrproject-rtos/zephyr/wiki/Development-Model#Maintaining_subsystem_branches and update them to the correct non-Gerrit workflow steps (or let me know what the steps should be and I can edit)

galak, 2018-02-13:
I updated to remove the subsystem stuff since we don't really have proper subsystem branches at this point.

galak, 2018-02-13:
@dbkinder can you look at the 'Flow3' link in Communication and Collaboration going nowhere.

dbkinder, 2018-02-21:
Typo in the link to this image is fixed.

closed 2018-02-21 by dbkinder