zephyrproject-rtos/zephyr #5807: Can't build Arduino_101 on Mac following instructions

jimmy-huang, 2018-01-24:
Using the 1.10.0 release, after I build crosstool-ng, I am unable to build the arduino_101 target on MacOS Sierra:

It first said that it can't find the toolchain:
-- Generating zephyr/include/generated/autoconf.h
CMake Error at /Users/zhuang1/projects/zephyrjs/zephyr.js/deps/zephyr/cmake/extensions.cmake:811 (message):
  No such file or directory: CMAKE_READELF:
  '/Volumes/CrossToolNG/x-tools/i586-zephyr-elfiamcu/bin/i586-zephyr-elfiamcu-readelf'

So it turns out that after building crosstool-ng, the tool chain is located under
/Volumes/CrossToolNG/x-tools/i586-zephyr-elf/i586-zephyr-elf

So I commented out this line at cmake/toolchain-xtools.cmake:
#set(CROSS_COMPILE ${TOOLCHAIN_HOME}/${CROSS_COMPILE_TARGET}/bin/${CROSS_COMPILE_TARGET}-)

then I I pass in the actual path:
cmake -DBOARD=arduino_101 -DCROSS_COMPILE=/Volumes/CrossToolNG/x-tools/i586-zephyr-elf/bin/i586-zephyr-elf- ..

then it fails to build with make:

[ 92%] Linking C executable zephyr_prebuilt.elf
/Volumes/CrossToolNG/x-tools/i586-zephyr-elf/lib/gcc/i586-zephyr-elf/6.3.0/../../../../i586-zephyr-elf/bin/ld: Warning: size of symbol `main' changed from 22 in ../libapp.a(main.c.obj) to 5 in kernel/libkernel.a(init.c.obj)
/Volumes/CrossToolNG/x-tools/i586-zephyr-elf/lib/gcc/i586-zephyr-elf/6.3.0/../../../../i586-zephyr-elf/bin/ld: skipping incompatible /Volumes/CrossToolNG/x-tools/i586-zephyr-elf/lib/gcc/i586-zephyr-elf/6.3.0/libgcc.a when searching for -lgcc
/Volumes/CrossToolNG/x-tools/i586-zephyr-elf/lib/gcc/i586-zephyr-elf/6.3.0/../../../../i586-zephyr-elf/bin/ld: skipping incompatible /Volumes/CrossToolNG/x-tools/i586-zephyr-elf/lib/gcc/i586-zephyr-elf/6.3.0/libgcc.a when searching for -lgcc
/Volumes/CrossToolNG/x-tools/i586-zephyr-elf/lib/gcc/i586-zephyr-elf/6.3.0/../../../../i586-zephyr-elf/bin/ld: cannot find -lgcc
collect2: error: ld returned 1 exit status
make[2]: *** [zephyr/zephyr_prebuilt.elf] Error 1
make[1]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
make: *** [all] Error 2

SebastianBoe, 2018-01-24:
Hi,

I think that when following the instructions a toolchain for i586 was built. But arduino_101 requires an iamcu toolchain.

At this step:

cp ${ZEPHYR_BASE}/scripts/cross_compiler/i586.config .config

try copying iamcu.config instead. See http://docs.zephyrproject.org/getting_started/installation_mac.html#setting-the-toolchain-options

SebastianBoe, 2018-01-24:
@dbkinder : Documentation might be ambiguous, example instructions were read as actual instructions:

http://docs.zephyrproject.org/getting_started/installation_mac.html#setting-the-toolchain-options

> Currently the following configurations are provided:
> 
> i586.config: for standard ABI, for example for Galileo and qemu_x86
> iamcu.config: for IAMCU ABI, for example for the Arduino 101
> nios2.config: for Nios II boards
> cp ${ZEPHYR_BASE}/scripts/cross_compiler/i586.config .config

jimmy-huang, 2018-01-24:
Yes, I did copy the iamcu.config and built that instead.  I use the i586.config to build QEMU, and I had similar issues, and I had to modify the cmake as well, to get that working.  I was going to file a bug for that as well, but eventually I figured out how to get that to build at least.  You can see the instructions including work-arounds we have to do to get QEMU working here, but A101 we are still not sure.

https://github.com/intel/zephyr.js/pull/1780

nashif, 2018-04-30:
duplicated by #6912

closed 2018-04-30 by nashif