zephyrproject-rtos/zephyr #5813: CII Badge: Default crytpo chooses weak algorithms

agross-linaro, 2018-01-24:
The default security mechanisms within the software produced by the project MUST NOT depend on cryptographic algorithms or modes with known serious weaknesses (e.g., the SHA-1 cryptographic hash algorithm or the CBC mode in SSH). 

We need to modify the current crypto samples to use algorithms with no known weaknesses.

nashif, 2018-02-24:
> We need to modify the current crypto samples to use algorithms with no known weaknesses.

I do not think this need to depend on samples, if we have some default system configuration that selects a weak crypto, then it should be changed, the samples and tests need to test all cryptos, also the weak ones.

So this should be seen as "Met" if we have no defaults in the system pointing to weak cryptos, and I do not think we have this, do we?

agross-linaro, 2019-02-20:
Samples have been added/modified to avoid non-weak crypto.  Specifically we have a google iot application that shows an example of how to properly implement TLS communications with a cloud service.

closed 2019-02-20 by agross-linaro