zephyrproject-rtos/zephyr #5817: socket.h: Using #define for POSIX redefinition of zsock_ functions has unintended consequences

GAnthony, 2018-01-25:
While porting mqtt to sockets, by including socket.h in mqtt.c, the socket.h definition:

`#define connect zsock_connect
`

breaks mqtt.c compilation by replacing ctx->connect with ctx->zsock_connect; however, zsock_connect is not a field of the struct mqtt_ctx  (but 'connect' is).

In general, this could be an issue with any function symbol re-definition made via #define.

A safer though more verbose approach which resolves the problem would be to use inline functions for the re-definitions:

```
static inline int connect(int sock,
			  const struct sockaddr *addr,
			  socklen_t addrlen)
{
	return zsock_connect(sock, addr, addrlen);
}
```

I log this as an issue to see if there are any other better suggestions.

I can workaround the problem for now by ensuring socket.h comes before all appearances of the symbol 'connect' in all headers and source files (the pre-processor effectively converting each instance consistently to 'zsock_connect').

pfalcon, 2018-01-26:
@GAnthony : Ack. Well, that shows why I'm conservative with not touching "connect" and friends by default - even if planned about, some issues still pop up (so not planning even worse ;-) ). Preprocessor defines were also selected as a conservative measure "working everywhere for decades", but I agree that the issue you present needs to be fixed, in the way suggested.

Let me know if you're interested to prepare a patch, or if you'd prefer that I did that.

GAnthony, 2018-01-26:
@pfalcon: I can do the patch.

PR #5852 (2018-01-30): net: sockets: Define POSIX function symbols as inline functions
Previously, POSIX function names were aliased to zsock_ function
names using, for example:

	#define connect zsock_connect

This caused the C preprocessor to replace any symbol named 'connect',
whether a function or not, in all source code which included socket.h,
with 'zsock_connect'.  This generated unintended code where the symbol
'connect' was used as the name of a structure field (as in mqtt.h).

This new inline definition is applied to all the POSIX function symbols,
with the exception of fcntl, a redefinition of which would [... cut]
commit f4e08b00 (2018-01-30): net: sockets: Define POSIX function symbols as inline functions

Previously, POSIX function names were aliased to zsock_ function
names using, for example:

	#define connect zsock_connect

This caused the C preprocessor to replace any symbol named 'connect',
whether a function or not, in all source code which included socket.h,
with 'zsock_connect'.  This generated unintended code where the symbol
'connect' was used as the name of a structure field (as in mqtt.h).

This new inline definition is applied to all the POSIX function symbols,
with the exception of fcntl, a redefinition of which woul [... cut]
closed 2018-01-30 by nashif