zephyrproject-rtos/zephyr #5821: [MSYS2] Unable to build Zephyr

SebastianBoe, 2018-01-25:
Hi,

following the installation instructions I was unable to build Zephyr using MSYS2.

The error I get is:

> sebo@SEBO-WIN MINGW64 ~/zephyr/samples/hello_world/build
> $ cmake -GNinja -DBOARD=nrf52_pca10040 ..
> CMake Deprecation Warning at C:/msys64/home/sebo/zephyr/cmake/app/boilerplate.cmake:37 (cmake_policy):
>   The OLD behavior for policy CMP0000 will be removed from a future version
>   of CMake.
> 
>   The cmake-policies(7) manual explains that the OLD behaviors of all
>   policies are deprecated and that a policy should be set to OLD only under
>   specific short-term circumstances.  Projects should be ported to the NEW
>   behavior and not rely on setting a policy to OLD.
> Call Stack (most recent call first):
>   CMakeLists.txt:2 (include)
> 
> 
> ############### C:/msys64/home/sebo/zephyr
> ############### C:/msys64/home/sebo/zephyr
> -- Selected BOARD nrf52_pca10040
> C:/msys64/home/sebo/zephyr/VERSION
> CMake Error at C:/msys64/home/sebo/zephyr/cmake/version.cmake:5 (file):
>   file failed to open for reading (No such file or directory):
> 
>     /home/sebo/zephyr/samples/hello_world/C:/msys64/home/sebo/zephyr/VERSION
> Call Stack (most recent call first):
>   C:/msys64/home/sebo/zephyr/cmake/app/boilerplate.cmake:220 (include)
>   CMakeLists.txt:2 (include)
> 
> 
> C:/msys64/home/sebo/zephyr/VERSION
> CMake Error at C:/msys64/home/sebo/zephyr/cmake/version.cmake:8 (string):
>   string sub-command REGEX, mode MATCH needs at least 5 arguments total to
>   command.
> Call Stack (most recent call first):
>   C:/msys64/home/sebo/zephyr/cmake/app/boilerplate.cmake:220 (include)
>   CMakeLists.txt:2 (include)
> 
> 
> CMake Error at C:/msys64/home/sebo/zephyr/cmake/version.cmake:11 (string):
>   string sub-command REGEX, mode MATCH needs at least 5 arguments total to
>   command.
> Call Stack (most recent call first):
>   C:/msys64/home/sebo/zephyr/cmake/app/boilerplate.cmake:220 (include)
>   CMakeLists.txt:2 (include)
> 
> 
> CMake Error at C:/msys64/home/sebo/zephyr/cmake/version.cmake:14 (string):
>   string sub-command REGEX, mode MATCH needs at least 5 arguments total to
>   command.
> Call Stack (most recent call first):
>   C:/msys64/home/sebo/zephyr/cmake/app/boilerplate.cmake:220 (include)
>   CMakeLists.txt:2 (include)
> 
> 
> CMake Error at C:/msys64/home/sebo/zephyr/cmake/version.cmake:17 (string):
>   string sub-command REGEX, mode MATCH needs at least 5 arguments total to
>   command.
> Call Stack (most recent call first):
>   C:/msys64/home/sebo/zephyr/cmake/app/boilerplate.cmake:220 (include)
>   CMakeLists.txt:2 (include)
> 
> 
> CMake Error at C:/msys64/home/sebo/zephyr/cmake/version.cmake:20 (string):
>   string sub-command REGEX, mode MATCH needs at least 5 arguments total to
>   command.
> Call Stack (most recent call first):
>   C:/msys64/home/sebo/zephyr/cmake/app/boilerplate.cmake:220 (include)
>   CMakeLists.txt:2 (include)
> 
> 
> Zephyr version: ..
> CMake Error at C:/msys64/home/sebo/zephyr/cmake/version.cmake:35 (math):
>   math cannot parse the expression: "( << 16) + ( << 8) + ()": syntax error,
>   unexpected exp_SHIFTLEFT, expecting exp_PLUS or exp_MINUS or exp_OPENPARENT
>   or exp_NUMBER (4)
> Call Stack (most recent call first):
>   C:/msys64/home/sebo/zephyr/cmake/app/boilerplate.cmake:220 (include)
>   CMakeLists.txt:2 (include)
> 
> 
> CMake Error at C:/msys64/home/sebo/zephyr/cmake/version.cmake:36 (math):
>   math cannot parse the expression: "( << 16) + ( << 8) + ()": syntax error,
>   unexpected exp_SHIFTLEFT, expecting exp_PLUS or exp_MINUS or exp_OPENPARENT
>   or exp_NUMBER (4)
> Call Stack (most recent call first):
>   C:/msys64/home/sebo/zephyr/cmake/app/boilerplate.cmake:220 (include)
>   CMakeLists.txt:2 (include)
> 
> 
> CMake Error at C:/msys64/home/sebo/zephyr/cmake/version.cmake:37 (math):
>   math cannot parse the expression: "( << 24) + ( << 16) + ( << 8) + ":
>   syntax error, unexpected exp_SHIFTLEFT, expecting exp_PLUS or exp_MINUS or
>   exp_OPENPARENT or exp_NUMBER (4)
> Call Stack (most recent call first):
>   C:/msys64/home/sebo/zephyr/cmake/app/boilerplate.cmake:220 (include)
>   CMakeLists.txt:2 (include)
> 
> 
> CMake Error at C:/msys64/home/sebo/zephyr/cmake/version.cmake:39 (to_hex):
>   to_hex Function invoked with incorrect arguments for function named: to_hex
> Call Stack (most recent call first):
>   C:/msys64/home/sebo/zephyr/cmake/app/boilerplate.cmake:220 (include)
>   CMakeLists.txt:2 (include)
> 
> 
> CMake Error at C:/msys64/home/sebo/zephyr/cmake/version.cmake:40 (to_hex):
>   to_hex Function invoked with incorrect arguments for function named: to_hex
> Call Stack (most recent call first):
>   C:/msys64/home/sebo/zephyr/cmake/app/boilerplate.cmake:220 (include)
>   CMakeLists.txt:2 (include)
> 
> 
> CMake Error at C:/msys64/home/sebo/zephyr/cmake/kconfig.cmake:93 (message):
>   File not found:
>   /home/sebo/zephyr/samples/hello_world/C:/msys64/home/sebo/zephyr/boards/arm/nrf52_pca10040/nrf52_pca10040_defconfig
> Call Stack (most recent call first):
>   C:/msys64/home/sebo/zephyr/cmake/app/boilerplate.cmake:222 (include)
>   CMakeLists.txt:2 (include)
> 
> 
> -- Configuring incomplete, errors occurred!
> 
> sebo@SEBO-WIN MINGW64 ~/zephyr/samples/hello_world/build
> $
> 

I added some debug prints to inspect the path that is corrupted:

message( "${ZEPHYR_BASE}/VERSION")
file(READ ${ZEPHYR_BASE}/VERSION ver)
message( "${ZEPHYR_BASE}/VERSION")

I can't explain that "file(READ [...])" is prefixing the file with "/home/sebo/zephyr/samples/hello_world/"

carlescufi, 2018-02-12:
You need to use a version of Ninja natively compile for MSYS2, I've updated the doc to reflect this.

carlescufi, 2018-02-12:
Also from your log it seems you are on a MinGW shell, which will not work. As described in the build instructions you need to open an MSYS2 MSYS shell.

PR #6146 (2018-02-12): doc: getting_started: Windows build doc fixes
Correct 2 items in the Windows Getting Started doc:

 * dtc-msys2 is now an upstream Chocolatey package
 * Add instructions to clone and build Ninja for MSYS2

Fixes #5808
Fixes #5821
closed 2018-02-12 by nashif