zephyrproject-rtos/zephyr #5833: Script to import mcux sdk

MaureenHelm, 2018-01-25:
New versions of the mcux sdk need to be imported whenever a new nxp soc is added to zephyr, or when nxp releases updated mcux drivers for an existing soc. Create a script to automate importing the mcux sdk into zephyr.

SebastianBoe, 2018-01-26:
For nrf's equivalent problem we intend to eventually re-use the mechanism used by OpenThread to import external source code. 

To solve this problem one might want to look into if using CMake's ExternalProject feature is powerful enough.

https://github.com/SebastianBoe/zephyr/blob/50e285cd6879da8941dab56d348ce222bc11dff0/subsys/net/lib/openthread/CMakeLists.txt

https://cmake.org/cmake/help/latest/module/ExternalProject.html

MaureenHelm, 2018-07-13:
Fixed by #8923

closed 2018-07-13 by MaureenHelm