zephyrproject-rtos/zephyr #5836: spi: stm32: convert remaining boards that support SPI to using dts

DWagenk, 2018-01-26:
As a follow up to #5582 we need to convert the stm32 boards, that already support SPI to using dts for SPI configuration. 
Right now building e.g. tests\drivers\spi\spi_loopback for nucleo_f091rc fails, because spi_ll_stm32.c driver relies on  SPI_BASE_ADDRESS, SPINAME, SPIIRQ and SPI_IRQ_PRI wich are not defined yet for this board.

I guess this affects the following boards (i generated this list by searching which boards are stm32 based and have a default pinmux for SPI in their pinmux.c):

- [x] 96b_carbon
- [ ] 96b_neonkey
- [x] disco_l475_iot1
- [ ] nucleo_f091rc
- [ ] nucleo_f334r8
- [ ] nucleo_f401re
- [ ] nucleo_l432kc
- [ ] nucleo_l476rg

erwango, 2018-01-26:
As part of  this issue, it would be good to populate .fixup files under dts rather than in boards.
(and move existing SPI .fixup parts)
Would contribute to #5707

DWagenk, 2018-01-29:
I've got a question regarding what to enable per default. Since dts.fixup is now moved to soc family level, there's not much work left for enabling SPI on these boards again.
What is the effect of
```
&spiX {
	status = "ok";
};
```
in dts? Right now it doesn't matter, if I add it to <board>.dts, as long as the  `CONFIG_SPI_X` is set, it works (as long as it doesn't use some undefined PINMUX definitions).
Should all `spiX` that are present on a board be populated with `status = "ok";` in boards dts file? or just the ones that are enabled by default when CONFIG_SPI is set?
I tried to just follow the way it is done for I2C, but it is not consistent there either.

ydamigos, 2018-01-29:
@DWagenk It enables the generation of the following defines:
```
#define ST_STM32_SPI_XXXXXXX_BASE_ADDRESS_0
#define ST_STM32_SPI_XXXXXXX_BASE_ADDRESS_1
#define ST_STM32_SPI_XXXXXXX_IRQ_0
#define ST_STM32_SPI_XXXXXXX_IRQ_0_PRIORITY
#define ST_STM32_SPI_XXXXXXX_LABEL
#define ST_STM32_SPI_XXXXXXX_BASE_ADDRESS
```
For now, we need to enable spiX both in boards dts file and in Kconfig file. Later, it should be done only in boards dts file.

PR #5865 introduce a new approach and checks if `CONFIG_SPIX_BASE_ADDRESS` is defined in the driver to enable SPIX device.

DWagenk, 2018-01-29:
OK, so for now I would add `status = "ok";` to dts for all SPIs that a board supports and then just activate one of them as defualt SPI in boards Kconfig.default.
Sounds good?

ydamigos, 2018-01-29:
Yes, the ones that the board enables in Kconfig.defconfig.

DWagenk, 2018-01-29:
OK, thanks for the clarification!

PR #5883 (2018-02-20): dts: spi: add more dts nodes and fixup defines, convert boards to using dts
This adds the remaining SPI related fixup defines to STM32 dts.fixup files at SOC family level. I added fixup defines for all SPIs that might be present on a SOC of that specific STM32 SOC family. Same for SPI dts-nodes.

Also I added SPI configuration via device tree to all STM32 Boards, that already have SPI in their pinmux.c. 
- Spi nodes in \<board\>.dts file are populated matching boards existing pinmux default configuration.
- Selecting SPI on those boards defaults to enabling SPI_1 in Kconfig. Other SPIs are dis [... cut]
commit 1f43e7b6 (2018-02-20): boards: arm: STM32 boards use DT to configure SPI

Configure SPI using DT for the following STM32 boards:

96b_neonkey
nucleo_f091rc
nucleo_f334r8
nucleo_f401re
nucleo_l432kc
nucleo_l476rg

SPI nodes in <board>.dts file are populated matching boards existing
pinmux default configuration and enabled in Kconfig.defconfig if
SPI is enabled.

For nucleo_l476rg board SPI2 and SPI3 nodes are not yet added, because
of missing pinmux defines.

Fixes #5836
closed 2018-02-20 by galak