zephyrproject-rtos/zephyr #5853: Using newlibc in a project breaks 'rom_report' and 'ram_report' targets.

GAnthony, 2018-01-27:
To reproduce, add
`CONFIG_NEWLIB_LIBC=y`
to samples/hello_world/prj.conf.

Building results in the following error:

```
build[master]> make rom_report
[...]
Scanning dependencies of target rom_report
Traceback (most recent call last):
  File "/home/gpitney/zephyr/scripts/footprint/size_report", line 420, in <module>
    base + '/', None)
  File "/home/gpitney/zephyr/scripts/footprint/size_report", line 194, in generate_target_memory_section
    symbols_paths = dict(load_symbols_and_paths(bin_nm, elf_file_abs, source_dir))
  File "/home/gpitney/zephyr/scripts/footprint/size_report", line 57, in load_symbols_and_paths
    processed_path = Path(path).relative_to(Path(path_to_strip))
  File "/usr/lib/python3.4/pathlib.py", line 806, in relative_to
    .format(str(self), str(formatted)))
ValueError: '/usr/src/debug/newlib/2.4.0-r0/newlib-2.4.0/newlib/libc/include/string.h' does not start with '/home/gpitney/zephyr'
zephyr/cmake/reports/CMakeFiles/rom_report.dir/build.make:57: recipe for target 'zephyr/cmake/reports/CMakeFiles/rom_report' failed
make[3]: *** [zephyr/cmake/reports/CMakeFiles/rom_report] Error 1
CMakeFiles/Makefile2:2026: recipe for target 'zephyr/cmake/reports/CMakeFiles/rom_report.dir/all' failed
make[2]: *** [zephyr/cmake/reports/CMakeFiles/rom_report.dir/all] Error 2
CMakeFiles/Makefile2:2033: recipe for target 'zephyr/cmake/reports/CMakeFiles/rom_report.dir/rule' failed
make[1]: *** [zephyr/cmake/reports/CMakeFiles/rom_report.dir/rule] Error 2
Makefile:469: recipe for target 'rom_report' failed
make: *** [rom_report] Error 2
```

nashif, 2018-01-29:
duplicate of #5866, do not think it is related to newlib though

nashif, 2018-01-29:
should be fixed by #5878

closed 2018-01-29 by nashif

GAnthony, 2018-01-29:
Yes, that fixed it.  Thanks!