zephyrproject-rtos/zephyr #5857: net: TCP retransmit queue implementation is completely broken and works by a chance (when nobody's looking)

pfalcon, 2018-01-28:
Rexmit is implemented as `sys_snode_t sent_list;`, and may be accessed concurrently at least by net_tcp_send_data() and net_tcp_ack_received(), but slist.h is ridden with warnings like: "Note: the loop is unsafe and thus __cn should not be detached" and "This and other sys_slist_*() functions are not thread safe.".

Here's typical logging for when working by a chance doesn't work:

~~~
got send_pkt: 0x00408338
appended 1079
[net/tcp] [DBG] net_tcp_queue_data: (0x004064a0): [0x00405ee0] Queue 0x00408338 len 1079
[net/tcp] [DBG] net_tcp_trace: (0x+ack_rcv
-ack_rcv
004064a0): [0x00405ee0] pkt 0x00408338 src 8080 dst 51352 seq 0x9d70a89c (2641406108) ack 0x5db6e7bb (1572267963/0) flags uAPrsf win 1280 chk 0xd46c
prepare_segment:ok
[net/tcp] [DBG] net_tcp_queue_data: (0x004064a0): [0x00405ee0] ref pkt 0x00408338 new ref 2 (net_tcp_queue_data:758)
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Sending pkt 0x00408338 (1119 bytes)
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x00+ack_rcv
-ack_rcv
4064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408338 because it was already sent.
[... 240 lines cut ...]
[net/tcp] [DBG] tcp_retry_expired: (0x00406650): retry 2: [0x00405ee0] sent pkt 0x00408338
+ack_rcv
-ack_rcv
[net/tcp] [DBG] tcp_retry_expired: (0x00406650): [0x00405ee0] ref pkt 0x00408338 new ref 2 (tcp_retry_expired:188)
[net/tcp] [DBG] tcp_retry_expired: (0x00406650): retry 3: [0x00405ee0] sent pkt 0x00408338
+ack_rcv
-ack_rcv
[net/tcp] [DBG] tcp_retry_expired: (0x00406650): [0x00405ee0] ref pkt 0x00408338 new ref 2 (tcp_retry_expired:188)
[net/tcp] [DBG] tcp_retry_expired: (0x00406650): retry 4: [0x00405ee0] sent pkt 0x00408338
+ack_rcv
-ack_rcv
[net/tcp] [DBG] tcp_retry_expired: (0x00406650): [0x00405ee0] ref pkt 0x00408338 new ref 2 (tcp_retry_expired:188)
[net/tcp] [DBG] tcp_retry_expired: (0x00406650): retry 5: [0x00405ee0] sent pkt 0x00408338
+ack_rcv
-ack_rcv
[net/tcp] [DBG] tcp_retry_expired: (0x00406650): [0x00405ee0] ref pkt 0x00408338 new ref 2 (tcp_retry_expired:188)
[net/tcp] [DBG] tcp_retry_expired: (0x00406650): retry 6: [0x00405ee0] sent pkt 0x00408338
+ack_rcv
-ack_rcv
[net/tcp] [DBG] tcp_retry_expired: (0x00406650): [0x00405ee0] ref pkt 0x00408338 new ref 2 (tcp_retry_expired:188)
[net/tcp] [DBG] tcp_retry_expired: (0x00406650): retry 7: [0x00405ee0] sent pkt 0x00408338
+ack_rcv
-ack_rcv
-snd_dt
~~~

pfalcon, 2018-01-28:
Needless to say, as you start to enable logging or add printk's, stuff gets nicely serialized and problems go away. After removing one of having-made-perfect printk's:

~~~
+snd_dt
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Sending pkt 0x00408260 (1492 bytes)
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408260 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408260 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408260 because it was already sent.
[net/tcp] [DBG] net_tcp_send_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408260 because it was already sent.
[net/tcp] [DBG] net_tcp_send+ack_rcv
unrf(0x00408260)
unrf(0x00408260)
[net/pkt] [ERR] net_pkt_unref_debug: *** ERROR *** pkt 0x00408260 is freed already by net_tcp_ack_received():970 (net_tcp_ack_received():970)
-ack_rcv
_data: (0x004064a0): [0x00405ee0] Skipping pkt 0x00408260 because it was already sent.
sock_write(4215872, 489a5, 1079)
~~~

laperie, 2018-02-13:
@pfalcon , do you have any suggestions on how to fix that?

pfalcon, 2018-02-13:
> do you have any suggestions on how to fix that?

Well, this is a complex issue, which is hard to debug due to time-sensitivity to debugging, as shown above. So, ideas to approach it:

1. Find a way to do logging in a way to minimize affecting time-sensitive code, e.g. by adopting/extending @nashif's https://github.com/zephyrproject-rtos/zephyr/pull/5586 , or by some alternative.
2. Fix cases of avoidable extra allocation (== potential failure cases). I have some ideas, will file them later.
3. It's clear that there're access synchronization issues to rexmit queue, ideas of wrapping it with semaphore/mutax can be tried even independently of p.1.
4. Hopefully after all of the above, the general issue will be more clear (or maybe even resolved). Worst-case, we may need to rework TCP processing to be more synchronous (== conflict-free), rather than being done by competing, conflicting threads.
5. Actually, first of all above, there should be good reproduction cases (most of issues I face happen with out-of-tree projects). Working on that.

Given the current situation (1.11 release, sockets work to cover TLS), this is in backlog, but I'd really like to focus on this with 1.12 window opening. Marking the issue accordingly. (@rveerama1 also touched some of issues with TCP re-xmit in https://github.com/zephyrproject-rtos/zephyr/pull/6058, but I doubt that resolved everything, due p.3 above).

Of course, I'd appreciate other developers to triage this issue.

Thanks.

nashif, 2018-02-20:
is this already fixed? can someone please verify!

pfalcon, 2018-02-20:
> is this already fixed? can someone please verify!

It's not, just last week I was getting non-deterministic lockups and errors when playing with sockets code. Per my comment above and metadata on the issue, I'm going to dig into this when 1.12 windows opens.

nashif, 2018-05-15:
@jukka any news on this?

pfalcon, 2018-05-15:
> @jukka any news on this?

This is assigned to me, and I was working on streamlined ways to reproduce it - the original logs above were produced with a server written in MicroPython, trying to reproduce it with in-tree samples. But moving in that direction, I faced other issues, like https://github.com/zephyrproject-rtos/zephyr/pull/7476 , so this is backlogged somewhat. I also started to analyze/trace the source code trying for find issues with "static analysis".

jukka, 2018-05-15:
@jukkar

pfalcon, 2018-05-15:
@jukka: Oops, sorry for disturbing!

nashif, 2018-05-29:
@pfalcon What is the status? Are we going to see a fix for 1.12?

rljordan, 2019-03-25:
Any new updates on this? Does the issue still exist?

nashif, 2020-03-11:
No updates for almost 2 years, does not seem important for anyone? @jukkar

pfalcon, 2020-03-11:
The issues described here are (IMHO) the root cause of things like https://github.com/zephyrproject-rtos/zephyr/issues/23302. It seems like more people only now start to play with Zephyr networking and raise concerns.

jukkar, 2020-03-11:
I fixed several issues for TCP in #23334 that might actually help with this issue.

jukkar, 2020-04-28:
Old issue and things have improved a lot in the latest version. We are also migrating to new TCP version and the testing should go there. As agreed in release readiness and bug triage meeting, closing this issue.

closed 2020-04-28 by jukkar