zephyrproject-rtos/zephyr #5881: enabling THREAD_MONITOR causes tests to fail

nashif, 2018-01-29:
I was trying something where I needed THREAD_MONITOR enabled, this caused the test to fail by just enabling the option without any code changes.

To reproduce:
``
$ cd tests/kernel/semaphore/sema_api
$ echo "CONFIG_THREAD_MONITOR=y" >> prj.conf
# build and run under qemu
```

Results:
```
***** BOOTING ZEPHYR OS v1.10.99 - BUILD: Jan 29 2018 20:18:28 *****
Running test suite test_sema_api
===================================================================
starting test - test_sema_thread2thread
0x00412098 is not a valid k_sem
FATAL: syscall _handler_k_sem_init failed check: access denied
***** Kernel OOPS! *****
Current thread ID = 0x004120c0
Faulting segment:address = 0x002b:0x00001370
eax: 0x00000000, ebx: 0x00000000, ecx: 0x00000000, edx: 0x00000000
esi: 0x00000000, edi: 0x00000000, ebp: 0x00000000, esp: 0x004083b8
eflags: 0x246
Fatal fault in thread 0x004120c0! Aborting.
QEMU: Terminated
```

lpereira, 2018-01-29:
Requires CONFIG_OBJECT_TRACING=y in order to reproduce.

andrewboie, 2018-01-29:
I still cannot reproduce this. The test passes for me.
My prj.conf:

```
CONFIG_ZTEST=y
CONFIG_IRQ_OFFLOAD=y
CONFIG_THREAD_MONITOR=y
CONFIG_OBJECT_TRACING=y
```

I verified these are all getting set in the generated .config

nashif, 2018-01-29:
looks like it is a build system issue, @SebastianBoe , can you take a look?
if I first set configure in THREAD_MONITOR, it works. If I add OBJECT_TRACING=y and rebuild without cleaning, it fails, so something is off with kobject generation.

SebastianBoe, 2018-01-30:
I think I have found the root-cause will push a fix when I have it.

PR #5920 (2018-01-31): cmake: Fix dependencies between kobject_hash files
CMake's custom_command support is not great. It is technically
supported, but it is not easy to get it right.

As explained in "5. File-level dependencies of custom targets are not
propagated" from http://bit.ly/2GvwwEy. When a custom_command uses a
custom_target as input, the custom_command must DEPEND on both the
custom_target and the underlying file.

Longterm we need to do something more sofisticated to prevent these
issues from popping up. Like add some static analysis to detect badly
written CMake code or introduce an abstraction for cus [... cut]
commit f5758b5d (2018-01-31): cmake: Fix dependencies between kobject_hash files

CMake's custom_command support is not great. It is technically
supported, but it is not easy to get it right.

As explained in "5. File-level dependencies of custom targets are not
propagated" from http://bit.ly/2GvwwEy. When a custom_command uses a
custom_target as input, the custom_command must DEPEND on both the
custom_target and the underlying file.

Longterm we need to do something more sofisticated to prevent these
issues from popping up. Like add some static analysis to detect badly
written CMake code or introduce an abstraction for cu [... cut]
closed 2018-01-31 by nashif