zephyrproject-rtos/zephyr #5887: [Coverity CID: 182597] Control flow issues in /drivers/sensor/vl53l0x/vl53l0x.c

Shilpashree3lc, 2018-01-30:
Static code scan issues seen in File: /drivers/sensor/vl53l0x/vl53l0x.c
 Category: Control flow issues
 Function: vl53l0x_init
 Component: Drivers
 CID: 182597
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

Shilpashree3lc, 2018-03-06:
*** CID 182597:  Control flow issues  (NO_EFFECT)
/drivers/sensor/vl53l0x/vl53l0x.c: 272 in vl53l0x_init()
266     	if (ret < 0) {
267     		SYS_LOG_ERR("VL53L0X_DataInit return error (%d)", ret);
268     		return -ENOTSUP;
269     	}
270     
271     	ret = vl53l0x_setup_single_shot(dev);
>>>     CID 182597:  Control flow issues  (NO_EFFECT)
>>>     This less-than-zero comparison of an unsigned value is never true. "ret < 0".

272     	if (ret < 0) {
273     		return -ENOTSUP;
274     	}
275     
276     	dev->driver_api = &vl53l0x_api_funcs;
277     	return 0;

galak, 2018-03-06:
Fixed by #6250

closed 2018-03-06 by galak