zephyrproject-rtos/zephyr #5896: Accidentally using MSYS's python from native windows leads to obscure error messages

SebastianBoe, 2018-01-30:
If you are so unlucky as to have python3.exe from MSYS in your PATH and attempt a native Windows build you will observe an obscure failure that looks something like this:

```
-- Found PythonInterp: C:/msys64/usr/bin/python3.exe (found suitable version "3.6.2", minimum required is "3
-- Selected BOARD nrf52_pca10040
Zephyr version: 1.10.99
Parsing Kconfig tree in c:/work/zephyr/zephyr/Kconfig
Using C:/work/zephyr/zephyr/boards/arm/nrf52_pca10040/nrf52_pca10040_defconfig as base
Merging C:/work/zephyr/zephyr/samples/hello_world/prj.conf
-- Generating zephyr/include/generated/generated_dts_board.h
Traceback (most recent call last):
  File "c:/work/zephyr/zephyr/scripts/dts/extract_dts_includes.py", line 906, in <module>
    main()
  File "c:/work/zephyr/zephyr/scripts/dts/extract_dts_includes.py", line 833, in main
    yaml_list[c] = yaml.load(yf, Loader)
  File "/home/paza/.local/lib/python3.6/site-packages/yaml/__init__.py", line 72, in load
    return loader.get_single_data()
  File "/home/paza/.local/lib/python3.6/site-packages/yaml/constructor.py", line 37, in get_single_data
    return self.construct_document(node)
  File "/home/paza/.local/lib/python3.6/site-packages/yaml/constructor.py", line 46, in construct_document
    for dummy in generator:
  File "/home/paza/.local/lib/python3.6/site-packages/yaml/constructor.py", line 398, in construct_yaml_map
    value = self.construct_mapping(node)
  File "/home/paza/.local/lib/python3.6/site-packages/yaml/constructor.py", line 204, in construct_mapping
    return super().construct_mapping(node, deep=deep)
  File "/home/paza/.local/lib/python3.6/site-packages/yaml/constructor.py", line 129, in construct_mapping
    value = self.construct_object(value_node, deep=deep)
  File "/home/paza/.local/lib/python3.6/site-packages/yaml/constructor.py", line 86, in construct_object
    data = constructor(self, node)
  File "c:/work/zephyr/zephyr/scripts/dts/extract_dts_includes.py", line 117, in include
    return self.extractFile(self.construct_scalar(node))
  File "c:/work/zephyr/zephyr/scripts/dts/extract_dts_includes.py", line 145, in extractFile
    with open(filepath, 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/:/work/zephyr/zephyr/dts/bindings'
CMake Error at C:/work/zephyr/zephyr/dts/dts.cmake:111 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  C:/work/zephyr/zephyr/cmake/app/boilerplate.cmake:227 (include)
  CMakeLists.txt:2 (include)

-- Configuring incomplete, errors occurred!

```

The key part is:

`FileNotFoundError: [Errno 2] No such file or directory: '/:/work/zephyr/zephyr/dts/bindings'`

we should find a way to detect and give a sane error message for this misconfiguration.

SebastianBoe, 2018-01-30:
Perhaps we could look for the keyword msys64 in the python path for native-windows builds.

carlescufi, 2018-03-13:
@SebastianBoe is this really an issue? now that we have native builds our users should not install MSYS2 anyway

SebastianBoe, 2018-03-13:
MSYS2 is still supported IIRC.

carlescufi, 2018-03-13:
It is, but not supposed to be mixed with native builds :) That said it's true, we can fix this in the best way possible.

SebastianBoe, 2018-03-13:
Ah, my mistake, what this issue is saying, is that for a native build, if you happened to have MSYS2 installed allready, perhaps from some other project, the build system might misbehave, so it is not related to whether we support MSYS2 or not.

carlescufi, 2018-03-13:
And I wonder if we really should have provisions for this, because at the end of the day this perhaps beyond the scope of the Zephyr build system.

nashif, 2018-07-25:
not really a bug

closed 2018-07-25 by nashif