zephyrproject-rtos/zephyr #5941: native_posix: No support for "UART console"

pfalcon, 2018-02-01:
"UART" is the default implementation console handler for Zephyr. Currently, building any sample in samples/subsys/console/ for BOARD=native_posix fails:

samples/subsys/console/echo:
~~~
/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/subsys/console/getchar.c: In function ‘console_init’:
/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/subsys/console/getchar.c:116:32: error: ‘CONFIG_UART_CONSOLE_ON_DEV_NAME’ undeclared (first use in this function)
  uart_dev = device_get_binding(CONFIG_UART_CONSOLE_ON_DEV_NAME);
~~~

samples/subsys/console/getchar:
~~~
/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/subsys/console/getchar.c: In function ‘console_init’:
/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/subsys/console/getchar.c:116:32: error: ‘CONFIG_UART_CONSOLE_ON_DEV_NAME’ undeclared (first use in this function)
  uart_dev = device_get_binding(CONFIG_UART_CONSOLE_ON_DEV_NAME);
                                ^
~~~

It would be nice to get that working (and shouldn't be too hard - this "uart console" would just proxy reads/writes from POSIX console).

pfalcon, 2018-02-01:
@aescolar-ot: Can you please look at this?

This got to be related to https://github.com/zephyrproject-rtos/zephyr/pull/5861, if you got shell working, that you must have got console working first, so I'm going to try/look into that patch next.

pfalcon, 2018-02-01:
So, the immediate problem here isn't "UART console" emulation, but that there's no "pull console" support for native_port. Proposed patch is posted as https://github.com/zephyrproject-rtos/zephyr/pull/5947

galak, 2018-12-06:
Is this still relevant?

pfalcon, 2018-12-06:
Yes, still relevant. However, recent console/tty patches may offer an alternative way to implement it for native_posix, simply bypass UART layer. But low priority, not sure that work fits in 1.14 timeframe.

Ref https://github.com/zephyrproject-rtos/zephyr/issues/10512, which discusses that console should work with other underlying devices besides UART.

aescolar, 2019-01-16:
This is really a new feature/enhancement request, not a bug. Also not too high priority.
My understanding:
This subsys/console, how it is built now, it works only with a UART as backend, and only with those that support `CONFIG_UART_INTERRUPT_DRIVEN`
The UART driver for native_posix was just the minimum to get the new shell subsystem working (so no UART_INTERRUPT_DRIVEN support).
One possible solution is to add that support to the native_posix UART driver. That should also enable other use cases for the UART in native_posix. But doing this, does not just take 30 minutes.
Another option would be to change this subsys/console to not assume a UART as backend, and build a very slim backend for native_posix that just plugs to the host getchar. This would probably be faster and easier.
=> I unassign myself. If somebody has the time/will to take it they can.

aescolar, 2019-05-22:
As just discussed in today's TSC, this is not big enough to qualify as feature request, but just as enhancement

closed 2024-02-08 by aescolar