zephyrproject-rtos/zephyr #5943: OT: utilsFlashWrite does not take into account the write-block-size

jfischer-phytec-iot, 2018-02-01:
OpenThread utilsFlashWrite does not take into account the write-block-size. OT uses a multiple of 4 while writing to the flash. This leads to a bus fault on a platform like K64F where write-block-size is 8.

Is it possible to abstract utilsFlash* over a file system like nffs?

KaSroka, 2018-02-02:
File abstraction can be used to emulate flash for OpenThread - as example you can see posix OpenThread platform: https://github.com/openthread/openthread/blob/master/examples/platforms/posix/flash.c 

We can also write our own settings modulethat uses file directly instead of flash: https://github.com/openthread/openthread/blob/master/examples/platforms/utils/settings.cpp

carlescufi, 2018-02-06:
@jfischer-phytec-iot Sounds to me like we could write an abstraction over FCB or NFFS. Any takers to implement this in the 1.11 timeline?

carlescufi, 2018-02-13:
@jfischer-phytec-iot are you going to give this a go?

nashif, 2018-05-15:
any update on this?

rlubos, 2019-02-25:
I'll look into this issue. We need a better way to store OpenThread settings - it became an recurring problem recently (#12944, #13687).

rlubos, 2020-01-17:
No longer an issue with #21796.

closed 2020-01-17 by rlubos