zephyrproject-rtos/zephyr #5953: Build system: typedef not fit with zephyr type when CONFIG_NEWLIB_LIBC is enabled

miyatsu, 2018-02-02:
The original type of "uint32_t" is "unsigned int" defined in minimal libc, stdint.h.
However, when link with new libc, the type of "uint32_t" is "long unsigned int" instead.

Some of the sample warning output are shown as follow.
```
note: expected ‘gpio_callback_handler_t {aka void (*)(struct device *, struct gpio_callback *, unsigned int)}’ but argument is of type ‘void (*)(struct device *, struct gpio_callback *, uint32_t) {aka void (*)(struct device *, struct gpio_callback *, long unsigned int)}’
```

pfalcon, 2018-02-05:
@miyatsu, Zephyr uses its own native types `u32_t`, `u16_t`, etc. They were introduced exactly to avoid issues like you describe. You would need to use them when writing Zephyr apps. And when interfacing with 3rd-party code, you may need to use explicit casts.

galak, 2018-02-06:
Can you provide a test/example that shows this issue.

closed 2018-04-19 by miyatsu