zephyrproject-rtos/zephyr #5956: samples/net/coap_server: Failed to send response to coap client

tarunkum, 2018-02-02:
I was trying to test coap_server along with coap_client with two different frdm_k64f boards.
The steps I have followed for coap_server & coap_cleint is as below:

cd zephyr/samples/net/coap_client
mkdir build && cd build && cmake -GNinja -DBOARD=frdm_k64f ../
ninja
cp -vr outdir/frdm_k64f/zephyr.bin /run/media/MBED/

cd zephyr/samples/net/coap_server
mkdir build && cd build && cmake -GNinja -DBOARD=frdm_k64f ../
ninja
cp -vr outdir/frdm_k64f/zephyr.bin /run/media/MBED1/

- Here I am trying make one frdm_k64f as coap_client & other one as coap_server.

- I have not done any changes in prj.conf files in both sample apps.

- After flashing both codes not able to see any response from coap server to coap_client console.

I have tested on latest commit today 
Details :- commit ec1b1df5edabc771ad7e15aba7515c607a6e5728

jukkar, 2018-02-13:
This one is still pending as I have only one frdm-k64f board available so I cannot re-produce this issue.

rveerama1, 2018-04-27:
@tarunkum when you made changes to samples/net/coap_client (GET, PUT and POST methods), how did you test that?, what was your counter part of CoAP server?

tarunkum, 2018-05-03:
@rveerama1 This issue can be closed. COAP is working fine now.