zephyrproject-rtos/zephyr #5958: "Ninja flash" swallows user prompts

mbolivar, 2018-02-02:
I have a problem using 'ninja flash' when I have multiple boards of the same type plugged in (multiple PyOCD boards, etc.). The issue is that the flasher scripts prompt the user for which board to flash from a list, but ninja swallows the prompt, so the command hangs forever.

This is a regression from "make flash".

mbolivar, 2018-02-02:
@carlescufi @SebastianBoe FYI

carlescufi, 2018-02-02:
I think the issue is CMake, since ninja uses CMake under the hood to invoke the flasher script.

```
$ ninja flash
[1/82] Generating always_rebuild
Building for board nrf51_pca10028
[2/2] Flashing nrf51_pca10028
FAILED: zephyr/cmake/flash/CMakeFiles/flash
cd /home/carles/src/zephyr/samples/bluetooth/beacon/b && /usr/bin/cmake -E env /usr/bin/python3 /home/carles/src/zephyr/scripts/support/zephyr_flash_debug.py nrfjprog flash --board-dir=/home/carles/src/zephyr/boards/arm/nrf51_pca10028 --kernel-elf=/home/carles/src/zephyr/samples/bluetooth/beacon/b/zephyr/zephyr.elf --kernel-hex=/home/carles/src/zephyr/samples/bluetooth/beacon/b/zephyr/zephyr.hex --kernel-bin=/home/carles/src/zephyr/samples/bluetooth/beacon/b/zephyr/zephyr.bin --gdb=/home/carles/prog/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gdb --openocd=/home/carles/prog/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd --openocd-search=/home/carles/prog/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts --nrf-family=NRF51
There are multiple boards connected.
1. 681089210
2. 681726921
Please select one with desired serial number (1-2): Error: EOF when reading a line
(Re-run as "/home/carles/src/zephyr/scripts/support/zephyr_flash_debug.py --verbose nrfjprog ..." or set CMAKE_VERBOSE_MAKEFILE for a stack trace.)
ninja: build stopped: subcommand failed.
```

mbolivar, 2018-02-02:
I can reproduce that behavior with nrfjprog. The behavior with PyOCD is different but still a problem:

```
$ ninja flash
[1/84] Generating always_rebuild
Building for board nrf52_blenano2
[1/2] Flashing nrf52_blenano2
```

This hangs forever with two boards plugged in. Compare that with 'make flash', which forwards the prompt to the user printed by pyocd-flashtool:

```
$ make flash
[  1%] Generating always_rebuild
Building for board nrf52_blenano2
[  1%] Generating misc/generated/syscalls.json
[  2%] Built target syscall_list_h_target
[  3%] Built target syscall_macros_h_target
[  5%] Built target offsets
[  6%] Built target offsets_h
[ 27%] Built target kernel
[ 37%] Built target app
[ 38%] Built target linker_pass2_script
[ 71%] Built target zephyr
[ 72%] Built target linker_script
[ 73%] Built target subsys__bluetooth__common
[ 76%] Built target subsys__bluetooth__host
[ 91%] Built target subsys__bluetooth__controller
[ 93%] Built target subsys__net
[ 95%] Built target zephyr_prebuilt
[ 98%] Built target kernel_elf
Scanning dependencies of target flash
[100%] Flashing nrf52_blenano2
Flashing Target Device
id => usbinfo | boardname
0 => Unknown Board [nrf52]
1 => RedBearLab-BLE-Nano [nrf52]
input id num to choice your board want to connect
```

mbolivar, 2018-02-02:
> I think the issue is CMake, since ninja uses CMake under the hood to invoke the flasher script.

But why would it be CMake if it works with the Make backend, but doesn't work with Ninja, and no other changes are made?

mbolivar, 2018-02-02:
Potentially relevant: https://ninja-build.org/manual.html#_the_literal_console_literal_pool

carlescufi, 2018-02-02:
>But why would it be CMake if it works with the Make backend, but doesn't work with Ninja, and no other changes are made?

I was mistaken, apologies for the red herring

PR #5959 (2018-02-02): cmake: fix 'ninja flash' et al with multiple boards plugged in
Fixes: #5958 

Ninja by default will suppress command output and input. This is a
problem for the 'flash', 'debug', etc. targets, which may need to
interact with the user (and which display interactive output like
progress bars).

To fix this, add the USES_TERMINAL option to their target
definitions. This puts them in the console pool on Ninja, allowing the
commands to interact with the console.

References:

https://ninja-build.org/manual.html#_the_literal_console_literal_pool
https://cmake.org/cmake/help/v3.8/command/add_custom_t [... cut]
closed 2018-02-02 by carlescufi