zephyrproject-rtos/zephyr #5968: datastructure for LIFO

wenhuizhang, 2018-02-05:
Hi, to whom it may concern:

I don't understand why LIFO and FIFO are all initialized with a queue please?

https://github.com/zephyrproject-rtos/zephyr/blob/2251ea04a8eeef76f3bc4bbe58ea4132a2f67fad/include/kernel.h#L1937

```
struct k_lifo {
	struct k_queue _queue;
};

#define _K_LIFO_INITIALIZER(obj) \
	{ \
	._queue = _K_QUEUE_INITIALIZER(obj._queue) \
	}

#define K_LIFO_INITIALIZER DEPRECATED_MACRO _K_LIFO_INITIALIZER

```

nashif, 2018-02-07:
@Vudentz can you please look into this, AFAIK you made changes in this area.

Vudentz, 2018-02-08:
Well they both are using a queue as storage for the items, in fact both k_fifo and k_lifo could be replaced by k_queue, but initially lifo and fifo did not share a common storage so there was a lot of code duplication.

closed 2018-02-14 by Vudentz